arendst/Tasmota #5068: Setting up GIPO pins via user_config

bechtold, 2019-01-29:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : PlatformIO
- [x] Hardware used : Wemos D1 mini
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
Not necessary I guess.
```

I am trying to setup a Wems D1 mini via user_config_override.h. Everythings works fine. WIFI, Module, MQTT. 
But I can't figure out how to setup the GIPO Pins.
Via Serial I can use 
```
GPIO2 7
```
to set the second pin (D4) to be a WS2812 Pin. But how do I set this up in the user_config?

Setting the module works fine like this:
```
#define MODULE                 WEMOS      // [Module] Select default model from sonoff_template.h
```

I tried the following in different version, but it didn't work:
```
#define GPIO2                  GPIO_WS2812
#define GPIO2                  7
#define GPIO2                  WS2812
#define G2                        7
```

Setting the GPIO Setting via Web or Serial is working fine. But I would like to set this via user_config. Any hint appreciated.

andrethomas, 2019-01-30:
This is no provision for this in my_user_config.h

andrethomas, 2019-01-30:
Every bit of flash counts but yes it will take a bit of coding to make that possible for I think 2 or 3 people in the last few months so not worth the time or the flash or the complexities emanating from users using the defines incorrectly.

If you configure the wifi settings once you can flash the devices with that.

After initial connect to wifi network you can send commands to it via http using a script.

wget http://192.168.42.2/cm?cmnd=topic%20mynewtopic as an example will issue command `topic mynewtopic` on the device with IP 192.168.42.2 - Not the %20 inserted to represent a space.

andrethomas, 2019-01-30:
That is also an option, remember you can use backlog command to issue all the commands in one line...

backlog topic mynewtopic;nextcommand;nextcommand;nextcommand;nextcommand