arendst/Tasmota #5069: TUYA SP10 support

bauerju, 2019-01-30:
Hi 
I got an TUYA SP10 plug with night light and energy monitor (BL0397) 
It is nearly the same like the KA10. To enable the light via website, I changed GPIO01 from GPIO_LED2 to 
GPIO_RELAY2.  in file sonoff_template.h

```
{ "SP10",         // TUYA SP10 (ESP8285 - BL0937 Energy Monitoring)
     0,                // GPIO00
     GPIO_REL2,        // GPIO01 Nachtlicht
     0,                // GPIO02
     GPIO_KEY1,        // GPIO03 Button
     GPIO_HJL_CF,      // GPIO04 BL0937 CF power
     GPIO_NRG_CF1,     // GPIO05 BL0937 CF1 voltage / current
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_NRG_SEL_INV, // GPIO12 BL0937 Sel output (1 = Voltage)
     GPIO_LED2,        // GPIO13 Blue LED - Power status
     GPIO_REL1,        // GPIO14 Relay 1, Red Led
     0, 0, 0
  },
```

netpok, 2019-01-30:
I have an S10 clone too, I will test it later but probably better to use GPIO01 as PWM1 in order to make it dimmable.

ascillato, 2019-01-30:
A dimmable plug? Sorry I don't think the hardware will Support that.

ascillato2, 2019-01-30:
Please, when you have time, add this information to the wiki. Thank you very much. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-01-30 by ascillato2

meingraham, 2019-01-30:
@ascillato - I believe it's actually two devices in one - a night light and a plug all in one unit. The night light is dimmable and the plug's on/off relay is controllable separately.