arendst/Tasmota #5070: TLS connection fails on Sonoff SV

andreas-bulling, 2019-01-30:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [] Development/Compiler/Upload tools used :
- [x] Hardware used : Sonoff SV
- [ ] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? : development
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
Hi, I can't get the TLS connection to work between a Sonoff SV flashed with the latest Tasmota and my Mosquito MQTT broker.

The console output shows 
```
MQT: Attempting connection...
19:02:03 MQT: Connect failed to home.fritz.box:8883, rc -5. Retry in 10 sec

and mosquitto.log
```
1548871418: New connection from 192.168.178.29 on port 8883.
1548871418: OpenSSL Error: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
1548871418: Socket error on client <unknown>, disconnecting.

I searched for hours online but couldn't find a fix for this.
```
```

ascillato2, 2019-01-31:
Hi, please, share your status 0

ascillato2, 2019-01-31:
Your config seems ok. So, if the RSSI is 100% you also have a perfect wifi signal. Have you tried to erase all flash with esptool.py as explained in the wiki for the device that don't work, and then flash the firmware again?

majherek, 2019-01-31:
@andreas-bulling Aye you use webserver and MQTT TLS in the same time?

andreas-bulling, 2019-01-31:
> Your config seems ok. So, if the RSSI is 100% you also have a perfect wifi signal. Have you tried to erase all flash with esptool.py as explained in the wiki for the device that don't work, and then flash the firmware again?

This I didn't do yet. Will give it a try and report back, thanks!

andreas-bulling, 2019-01-31:
> @andreas-bulling Aye you use webserver and MQTT TLS in the same time?

I've tried both - with and without. No difference :(

andreas-bulling, 2019-02-01:
I tried with esphomelib. First it didn't work but after setting `tls_version tlsv1` for mosquitto the node could successfully connect. Back to Tasmoto and things don't work any more...

andreas-bulling, 2019-02-01:
I switched to esphomelib completely. Has the additional benefit that I don't need mqtt anymore thanks to the esphomelib native home assistant interface

ascillato, 2019-02-02:
Sorry to hear that. Anyway, I agree that you need to use what best suits your needs.

ascillato2, 2019-02-02:
This issue was very weird due to in one hardware it works and in another it doesn't.
Anyway, if you want to try again in the future, please do not hesitate on asking to reopen this issue or asking anything else. 

One thing I was about to ask you was if the mqtt client name was different in both devices. It should to make it work. 

Hope that now it is working ok for you. Thanks for all the testings.

closed 2019-02-02 by ascillato2