arendst/Tasmota #5072: Share ideas : Home assistance custom component for Tasmota

hhaim, 2019-01-30:
I'm new to Tasmota and HASS but I've developed a few hass custom components that might help others and wanted to share them. 

1. Counter sensor (e.g for water meter using pulse)  -- the issue here that the counter become zero after power/reset, and there is no way to save it to flash without wear the flash. rule to send a diff is  not good beacuse you can lose one of the telemetry (default MQTT QoS is zero from Tasmota -> Broker)  and I wanted to be more accurate here. 

Using this custom 
[counter](https://github.com/hhaim/hass/blob/master/custom_components/sensor/tasmota_counter.py)

you can define a water meter like this:

```
- platform: tasmota_counter
    name: water_total
    stopic: water_out
    counter_id: 1
    max_valid_diff: 2000
    unit_of_measurement: 'l'
    icon: mdi:water-pump
    expire_after: 300
    value_template: "{{ (4885 + (value))|int }}"
```

and it will take care of counter (again without saving it to flash and no special tricks in tasmota ) 

2.   MCP23017  device that has a few mqtt indications (interrupts and status) . I've used this device to convert my old alarm to be smart alarm. Each binary sensor has an interrupt and status and both are used by this component 

[MCP23017 binary sensor](https://github.com/hhaim/hass/blob/master/custom_components/binary_sensor/mqtt_alarm.py)

This component simplifies the way to define an alarm system using this sensor 

**Example**
```
- platform: mqtt_tasmota
    name: alarm-s0
    index: '1'
    stopic: alarm
```

3. A **Switch** that sync automatically from the command status and tele  (hass support only one) 
It simplify the the way you define a switch, and there is no way to get out of sync with hass (no need to define it @ script startup)

[tasmota switch](https://github.com/hhaim/hass/blob/master/custom_components/switch/mqtt_tasmota.py)

Example:
```
- platform: mqtt_tasmota
        name: b0
        stopic: b0
```

There are more custom components but they are general to hass
 
The full configuration is here:
[more examples here](https://github.com/hhaim/hass/blob/master/configuration.yaml)

let me know what you think

andrethomas, 2019-01-30:
@hhaim 

There's an existing Wiki page for Home Assistant where you can post these if you are that way inclined.

https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

Thanks for sharing

hhaim, 2019-01-30:
Thanks, will incorporate that

closed 2019-01-30 by hhaim