arendst/Tasmota #5073: Ability to disable automatic calibration for MH-Z19B

oal, 2019-01-30:
**Have you look for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
Not really a problem, but it'd be nice to be able to disable automatic calibration for the MH-Z19B Co2 sensor.

**Describe the solution you'd like**
Support for the "Sensor15 1" command. It seems like the command is already there, but not enabled?
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_15_mhz19.ino#L80

But it's not handled in this switch statement:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_15_mhz19.ino#L275

**Describe alternatives you've considered**
I've considered trying to add an additional case to the switch statement mentioned above, but it looks too simple not to already have been added. Any ideas if more work is required?

**Additional context**
ESPEasy seems to support this, and uses the same packet of bytes for the command as the one already in the Sonoff-Tasmota source code: https://github.com/letscontrolit/ESPEasy/blob/b6507784828d2a92785f2027658e6d5df277ead2/src/_P049_MHZ19.ino#L66

andrethomas, 2019-01-30:
@oal 

In xsns_15_mhz19.ino (The driver file for the MHZ19) there is this line:

```
#define MHZ19_ABC_ENABLE             1       // Automatic Baseline Correction (0 = off, 1 = on (default))
```

If you set that to 0 before compiling it should disable automatic baseline correction.

Jason2866, 2019-01-30:
What is the use case for disable calibration?
I ask because i have some MHZ19B....

ascillato2, 2019-01-31:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

andrethomas, 2019-01-31:
> What is the use case for disable calibration?

Any area which does not reach 400ppm for at least 20 minutes within a 24 hour cycle... green house for example.

Jason2866, 2019-01-31:
@andrethomas 
mhh, i am sure that not all my sensors do reach 400ppm every day.
The do still work very well.

andrethomas, 2019-01-31:
@Jason2866 Mine does most of the time. It is not clear from the datasheet how they do it but it seems to be a 24 hour cycle - perhaps it has some hysteria built in insofar that it will not do anything unless there is a significant shift in measurements.

[image]

The one above was at 400 for a small period in time but I do not see any major shifts - I remember seeing it happen when I first installed it - after 24h the Co2 dropped 150ppm from the one telemetry period to the next and since then it has been similar to what is shown above.

oal, 2019-01-31:
Don't they set the lowest recorded value in the last 24 hours to be the 400 PPM baseline? So 400 may actually be a lot higher, if the lowest actual PPM in the previous 24 hours is >400PPM.

I was thinking about putting my sensors outside, perform calibration, then disable auto calibration. Wouldn't that give the most accurate readings? Unless they drift off and become inaccurate with auto calibration turned off.

andrethomas, 2019-01-31:
My observation is that it does that in a 24 hour cycle with

```
#define MHZ19_ABC_ENABLE             1
```

But if you do

```
#define MHZ19_ABC_ENABLE             0
```

It appears to not recalibrate unless you want to force it to recalibrate in which case you need to put it outdoors for at least 20 minutes and then issue command `sensor15 2` from the console but as I said the datasheet is not totally clear on this, just an observation.

andrethomas, 2019-01-31:
@oal compile time only at this point.

oal, 2019-01-31:
I'm not too familiar with the code base, but all that's needed is to add the `0` and `1` cases to the switch statement here? https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xsns_15_mhz19.ino#L275

The function that sends MH-Z19B commands seems to be able to handle that, right? Just making sure I'm not overlooking something obvious.

andrethomas, 2019-01-31:
I think that should do it, more or less - I'll look into adding it over the weekend as I'll need to find a space to store the value in settings.

Jason2866, 2019-01-31:
> Don't they set the lowest recorded value in the last 24 hours to be the 400 PPM baseline? So 400 may actually be a lot higher, if the lowest actual PPM in the previous 24 hours is >400PPM.
>
> I was [...]

My experience is the do work fine with autocalibration even the reach not 400ppm for a few days (weeks?)
400 ppm baseline do not shift! 
So that was my reason for asking for the use case switching off autocalibration

andrethomas, 2019-02-01:
@Jason2866 Yes, I also use it with it enabled but I will add one to the greenhouse at some stage so for that the automatic calibration will not work well and the only solution is to leave the sensor running outdoors with ABC disabled and then send command `Sensor15 2` to force calibration to 400ppm after it has been running outdoors for at least 20 minutes.

Either way, the default behaviour is not changed so it will still default to it being enabled.

Proposed PR https://github.com/arendst/Sonoff-Tasmota/pull/5088

andrethomas2, 2019-02-01:
Closing this issue as it the PR made for this request has been merged into development.

closed 2019-02-01 by andrethomas2