arendst/Tasmota #5075: Cannot flash - error 2 uploading (espcomm_sync failed) RF R2 Sonoff

vulemanijak, 2019-01-31:
Hi everyone, i just recently bought sonoff basic for the first time, and i was attempting to flash it but.. first it turned out to be new model with new layout, and it has esp8285 board on it. Couldn't find any youtube videos specificly for this new model, so i searched the internet and many tutorials, but no luck...
what i did when i was closest to flash:
1. Bought two TTL to USB adapters (first ft232rl than CP2102) [image]
[image]
2. Used VSCODE with PlatformIO plugin to setup software (Downloaded latest version of tasmota https://codeload.github.com/arendst/Sonoff-Tasmota/zip/v6.4.1)
3. Built using PlatformIO in VSC
4. Got fatal error that sonoff.ino.cpp is missing
5. Downloaded older version of tasmota witch contained it (https://codeload.github.com/arendst/Sonoff-Tasmota/zip/d527a53d65d7d4a267617e51cbf8586d71f9e4f9)
6. I had put sonoff.ino.cpp in latest tasmota version's sonoff folder.
7. Built successfuly
8. Connected Sonoff with first adapter (Wirings tripple checked)
9. tried to upload, got error 2 uploading (espcomm_sync failed)
10. tried various combinations of wiring and putting into flash mode ((TX-RX RX-TX / TX-TX RX-RX), (holding down button with all wires connected, putting usb in, releasing button),(Holding down button, putting usb in, putting 3v wire in, releasing button after few secs) in short every combination with both adapters.. also i tried in another COMs... none seemed to work, always getting same error...

Maybe there is different setup for esp8285 that i dont know of... just to mention, i tried to follow tedsluis comment in https://github.com/arendst/Sonoff-Tasmota/issues/2344, but no luck eather... hope someone will solve this... thanks!
*I am using windows 10
Sorry if my english is bad, its not my native language

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : ArduinoIDE, VSCode (PlatformIO), ESPeasy
- [x] Hardware used : Sonoff-Basic RF R2, CP2102, Windows 10, jump wires
- [x] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` : Error 2
```

```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-01-31:
Have you put gpio0 to gnd before powering the sonoff basic?

ascillato, 2019-01-31:
Remember that it is available the Tasmota Support Chat for addressing this type of issues in a faster way.

Github issues are meant for software bug reporting.

vulemanijak, 2019-01-31:
I tried putting gpio0(key?) to ground b4 powering few times, but read somewhere that pressing the button will do the work... so can i use some external source of power or should i buy new adapter? I saw many videos on yt with ft232rl working flawlessly

Jason2866, 2019-01-31:
Yes, it can work... 
Using a extern 3.3V power supply will solve your powering issue

vulemanijak, 2019-01-31:
Thanks, i will inform if it worked 😄

ascillato2, 2019-01-31:
Great.

Closing this issue as it is not a Tasmota software bug.

Anyway, please tell us here if that worked for you, when you have time to do the tests and if you need further assistant please do not hesitate on asking here or in the Tasmota Chat. Thanks.

Enjoy Tasmota :)

closed 2019-01-31 by ascillato2

vulemanijak, 2019-02-01:
@dawidkoldras i am trying to flash sonoff basic, gpio0 button is tall black button right? And which is power button? Maybe I didnt understand quite well sorry...

dawidkoldras, 2019-02-01:
@vulemanijak What i mean by power button is to deliver power to sonoff.

vulemanijak, 2019-02-01:
Oh ok, ill try that and get back here, thanks!

FrnBsts, 2019-02-15:
Exactly the same problem here trying to flash the new version of Sonoff with ESP8285.

· First I tried with the TTL to USB adapter ft232rl. I always obtained the espcomm_sync failed error. I couldn't see anything in the serial monitor.
· Then I tried to use a spare Arduino as a TTL to USB by connecting reset to ground. Now I can read the serial monitor, and it does enter to boot mode:(1,7). Following dawidkildras instructions I manage to enter boot mode:(1,0), but this gives me exactly the same error.
· I have tried with esptool,  PlatformIO and Arduino IDE and I get the same error.
· I have tried with two different Sonoff basic

Anyway, I still have doubts regarding the configuration to be choosen. I've tried multiple combinations, but I gess the correct one is generic ESP8285. I'm using board version 2.5.0 and the following options:

[image]

A final question is: do I need to hold the GPIO button the entire uploading process?

Thanks!!

dawidkoldras, 2019-02-15:
> Exactly the same problem here trying to flash the new version of Sonoff with ESP8285.
>
> · First I tried with the TTL to USB adapter ft232rl. I always obtained the espcomm_sync failed error. I [...]

I was holding GPIO button through entire process
[image]
Here are mine settings, if you enter boot mode:(1,0) it means that connection is correct.
Could you give me output from serial monitor, when you enter boot mode?
Another question, once you enter boot mode and it gives you all this informations as output, do serial monitor is stukk writing some signs like "&""%" etc. after that?

vulemanijak, 2019-02-15:
I have used arduino uno, holded jump wires douring the entire process at angle to make sure contact is good, and it worked flawlessly... watch newest drZzs tutorial, it should work, but use arduino uno to make sure you have enough curent to power sonoff

Jason2866, 2019-02-15:
Dont use FTDI adapters as written 100x times.... (Or just the ones with a EXTRA 3.3v power supply)
If flash fails it is 99% the USB serial adapter....

FrnBsts, 2019-02-18:
> I was holding GPIO button through entire process
> [image]
> Here are mine settings, if you enter boot mode:(1,0) it means that connection is correct.
> Could you give me output from serial monitor, when you enter boot mode?
> Another question, once you enter boot mode and it gives you all this informations as output, do serial monitor is still writing some signs like "&""%" etc. after that?

When I plug the current without pressing GPIO 0, I get this:
[image]

When I enter boot mode after pressing GPIO 0, only this line is prompted. The boot mode is normally (1,7), but after unplugging and plugging again the 3.3V it changes to (1,0). However, can I also get (1,#), where "#" is a number between 0 and 7.

[image]

I've been using Arduino Uno, pressing the GPIO 0 button the entire process and holding the cables to ensure a good contact, and I still get the same error :(

I have one more Sonoff that I haven't tried to flash, but I don't expect too much after having the same error in 2 of them.

Anyway, I am using the Esp 2.5.0 version in the board manager and tasmota 6.4.1. Maybe I could try chaging this, but again I don't think this is the problem.

FrnBsts, 2019-02-18:
Hey! Finally I managed to flash Tasmota to the same Sonoff as before. It seems that there was a problem with the cables I was using. I changed all of them and now it worked perfectly!

Sorry for all the troubles!!