arendst/Tasmota #5076: newbie doubt

jviure, 2019-01-31:
Hi people! I am going to change my sonoff default firmw to tasmota. Is posible to have inching  mode + interlock mode simultaneously? I want that.. if i click only R1, after X seconds turn-off auto (inching), but if I click R1, then click R2, R1 turn off auto (interlock). 

Is that possible? thanks! ;)

ascillato, 2019-01-31:
Yes. It is possible

Use command pulsetimer for making the relay to go off after some seconds. And interlock command for interlock. 

More info is on the wiki. 

If you need further assistant please find us in the Tasmota Support Chat. Thanks

jviure, 2019-01-31:
ok thanks, with the default firmware I think is not possible, right? well.. I will try tasmota!  thanks :)

ascillato, 2019-01-31:
AFAIK is not possible with the firmware from iTead.

ascillato, 2019-01-31:
If your question has been addressed, please close this issue. Thanks

jviure, 2019-01-31:
Thansk a lot, keep on going! ;)

jviure, 2019-02-01:
sorry, one more question, it's possible with tamota's firmware to connect (as with the default firmw) the remote control?  thanks!

netpok, 2019-02-01:
Sorry I misunderstood your question I thought you are asking about the remote control software for smartphones. RF remotes are using specific hardware and they are working with Tasmota. You can look for further info on the wiki: [Devices/Sonoff RF](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF)

jviure, 2019-02-01:
ok thanks netpok! so.. it's posible! great! thanks :) now I m thinking what software I can install.. domoticz,  openhab, hassio..?  too many solutions! jejej