arendst/Tasmota #5077: New Device: BHT-6000 Beca Smart Wifi Thermostat?

micgus, 2019-01-31:
Hello fellow Tasmotians.

I recently purchased a "Smart Wifi Thermostat" for my underfloor heating (altho available for other heating systems as well), and this type of thermostat is super common over here in the nordics.
Here's the product that i purchased but it seems to be available in a bunch of rebrands as well:

http://becaenergy.com/index.php/Home/New/productsinfo/id/52.html
https://www.aliexpress.com/item/BHT-6000-GBLW-LCD-Touch-Screen-Electric-Underfloor-Heating-Thermostat-Backlight-WIFI-16A-Works-with-Alexa/32865974596.html

I did search around to see if anyone had a go at it so far with their own firmware, but i couldn't find any info on it. 
Opened it up, and it is based on the TYWE3S with a SONiX SN8F57084 MCU, 
so reverse engineering it and getting Tasmota on it should be possible for the pros out there right? :)
Tried to eavesdrop on the serial communication between the TY and MCU but I'm not sure on where to get started on determining whats what in the communication.

[image]

[image]

[image]

Any one seen it before or perhaps are up for the challenge of getting tasmota on it?
Cheerio.

andrethomas, 2019-01-31:
The big question is the one you posted - what is in the serial data...

Normally the data would be in short bursts (packets/buckets) so you'd have to isolate those and then change values and see which bytes change and try and make sense of it.

Could be useful if you had a logic analyzer to dump the serial data but it is not impossible to do it with other serial monitor like tools (the name of which currently escapes me)

klausahrenberg, 2019-02-16:
I have a similar device: beca-002-wifi. It's possible to flash the ESP inside, but the ESP communicates with the Sonix-Controller, that means, the functionality itself can't be changed. But you can read and set all values. Actual I tested the MCU communication and will be able to provide a separate firmware for MQTT controlling within next week's. If you are interested to test your device with this, write me a comment.

micgus, 2019-02-16:
Oh, okay cool. I definitely want to test it out!
So far I have only mapped the serial commands but haven't had the time to start working on a firmware or tasmota integration.

Please share your fw when you are able to.

klausahrenberg, 2019-02-21:
Here is a first test version. 
[https://github.com/klausahrenberg/ThermostatBecaWifi](https://github.com/klausahrenberg/ThermostatBecaWifi)
Would good to know, if the MQTT state is correct on your device. All details you can find in the readme there.

micgus, 2019-02-21:
Looks great! I'll get to installing it after work today. 
Thanks!

baha00, 2019-12-03:
any update on this? I managed to load tasmota on it, with mock tuya cloud and load tasmota 7.0 but front buttons not reacting

lokidaibel, 2019-12-05:
Same here. With Tuya convert we can bring Tasmota easy on this device but what now ? Anyone has any clue of this ?

If you need any debug informations i have 8 Devices which can provide alot of infos.