arendst/Tasmota #5078: Commandline with request fails in Homeassistant

Caswell1000, 2019-01-31:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
- [x] Hardware used : 
- [x] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

I have a working sonoff basic, I use commandline on homeassistant to operate the sonoff using  :  ```command_on: "/usr/bin/curl -X POST http://something.duckdns.org:92/cm?user=admin&password=mypassword&cmnd=Power%20ON" ```

The above doesn't work on home assistant when password is activate but works when there is no password as follows:

```command_on: "/usr/bin/curl -X POST http://something.duckdns.org:92/cm?cmnd=Power%20ON" ```

The url with the password works when I implement it on a normal browser, but fails to operate on home assistant. the one without the password works in both normal browser and on Homeassistant. There is no error message directly related to this, it just does not switch on, the light switch returns to off position immediately with no action. I tried searching in vain for a solution.

The reason I need it to operate with a password is that I wand to be able to use it both outside and inside the homeassistant environment.

Jason2866, 2019-01-31:
It is not a issue in Tasmota since you wrote:
`The url with the password works when I implement it on a normal browser, but fails to operate on home assistant.`

andrethomas, 2019-01-31:
This works for me

`http://192.168.42.2/cm?user=admin&password=pass&cmnd=power%20ON``

andrethomas, 2019-01-31:
@Caswell1000 

wget is a linux command to download from a url to a file

So in this you would use this command from Linux command line:

`case wget -O /dev/null "http://192.168.42.2/cm?user=admin&password=pass&cmnd=Power%20ON"`

Not the URL is i inverted comma's because the & is an escape character to put a command into background processing from the Linux command line.

andrethomas, 2019-01-31:
So in hindsight looking at your original post you may need to do

```
command_on: "curl -X POST 'http://192.168.42.2/cm?user=admin&password=pass&cmnd=Power%20ON'"
```

You will note I enclosed the url in '

Caswell1000, 2019-01-31:
I was thinking the same after I saw the fascam example in "home assistant command line" write up, will try that next, I think the appersands (&) change things slightly, hence the one without passwords work because it does not have. Thanks will give feedback.