arendst/Tasmota #5081: Where to store large (12kb) settings item?

KonstantinMastak, 2019-01-31:
I want to add support of recording and playback of custom IR sequences as a lot of infared protocols are still not available in Tasmota. Since each "raw" IR sequence is up to 256 bytes (at least I hope this is enough in most cases) length and I want something like 50 slots for IR commands, I need to allocate about 13kb in settings array, or somethwere else in virtual EEPROM/real flash. 

I know that standard virtual EEPROM in ESP8266 is limited to 4096 bytes, and in current version of Tasmota only abbout 500 bytes left... On the other hand, I have still plenty of free flash in ESP8266 to allocate 12 kb I need.

So my question or request for advise is what is the best way to do it?

andrethomas, 2019-01-31:
Actually, if you're using a 1M firmware image on a 4M device you should be able to write to the unused part of the 4M flash which would theoretically start from sector 0x100 or address 0x1000000, but I have not tested this so you will need to do the testing.

KonstantinMastak, 2019-02-04:
Adrian,

Thank you very much for your detailed answer! It completely solves my
problem.
ᐧ