arendst/Tasmota #5082: Setup ESP8266-S1 On Oittm Aroma Essential Oil Diffuser

gridi88, 2019-02-01:
Hi everyone. I have flash correctly tasmota os on the aroma diffuser Oittm, and I am very happy. But I have difficulty to setup the various switch / botton / relay / led / etc options in the Generic Module to have full control of this fantastic accessory. I attach the link of the product, and maybe some photos
[image]
[image]
[image]
[image]

. I thank those who help me in advance. Thanks :) 
https://www.oittm.com/Oittm-Smart-Wi-fi-Aroma-Essential-Oil-Diffuser

ascillato, 2019-02-01:
Hi

Currently is not supported that device. Sorry. 

As it has other ICs also, may be some of the leds, relays, etc are managed by other mcu different than the esp8266. In that case, you will need to do some reverse engineering on that device for implementing Tasmota.

meingraham, 2019-05-06:
@gridi88  Potential solution or use of similar approach - https://blakadder.github.io/templates/oil_diffuser_550ml.html

tobias-d-oe, 2020-01-15:
@gridi88 Have you managed to get it working? I also have flashed one, but only can power on/off the device. Till now haven't managed to decode the TuyaReceived messages for resending. Every command seems to be fnID=11,dpId=1....

gridi88, 2020-01-15:
unfortunately I can't do anything else but turn it on and off.

> @gridi88 Have you managed to get it working? I also have flashed one, but only can power on/off the device. Till now haven't managed to decode the TuyaReceived messages for resending. Every command seems to be fnID=11,dpId=1....

unfortunately I can't do anything else but turn it on and off.

Staubgeborener, 2021-01-08:
@mrIngrid Anything new? Did you just flashed ` tasmota32.bin`? And how did you connected your cp2102 (or whatever this is) to the oittm? I see DBG1 and DBG2, but i could not find out the pinout for those.