arendst/Tasmota #5085: busch jaeger free@home

martin072, 2019-02-01:
This is a question about Busch & Jaeger Free@Home. 

I noticed that the KNX protocol was implemented a while ago. Not really familiar with this, but my newly build home will come with a  Busch & Jaeger Free@Home system. I was told this is a branch of their industrial KNX system. It uses both a Bus system or wifi (can be combined). I cannot find any information about technical spec's / protocols for Free@home, so I was wondering if this might be compatible with KNX or if there is a way to implement this within tasmota?

ascillato, 2019-02-01:
Tasmota supports KNX IP multicast protocol. So, as long as your devices also uses the same protocol, all should work fine.

martin072, 2019-02-01:
Understood, but as mentioned I cannot really find any information if KNX devices are compatible with Free@home (I am not able to test this as the house is still being built)

ascillato2, 2019-02-01:
As far as I could find using google, those are KNX devices.

closed 2019-02-01 by ascillato2