arendst/Tasmota #5087: Is this variant of MagicHome "Led SPI Controller" flashable ?

fabito, 2019-02-01:
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Hardware used : Magic Home Led SPI Controller

Hi, 

I just bought a MagicHome LED controller that seems different from the documented ones. It is very similar to [this one (reported in the espurna project)](https://github.com/xoseperez/espurna/issues/1437).
Check the pictures below:

[image]
[image]

I tried flasshing following the ["variant without rx/tx pads"](https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller#variant-without-rxtx-pads) instructions without success.
Apparently I can't boot in flash mode.

ascillato2, 2019-02-02:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-02 by ascillato2

RomeyRome206, 2019-11-22:
> If you solder all needed cables on ESP Module it will work!
> https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller#variant-without-rxtx-pads
> Connect Power Supply AFTER you have made already the connection GND to GPIO0

FWIW, that method doe mot work on this variant. TX/RX are linked to the ARM chip on the backside. I could only flash it by removing the ESP module from the board completely, then soldering it back on.

Chaot1, 2019-12-29:
> 
> 
> > If you solder all needed cables on ESP Module it will work!
> > https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller#variant-without-rxtx-pads
> > Connect Power Supply AFTER you have made already the connection GND to GPIO0
> 
> FWIW, that method doe not work on this variant. TX/RX are linked to the ARM chip on the backside. I could only flash it by removing the ESP module from the board completely, then soldering it back on.

I was able to flash tasmota on the item. But i can't find which GPIOs are assigned to the output.
Did you find it?

LordDuplo, 2020-01-11:
Hi,

i have the same controller.
I have flashed Tasmota on it with success.
But i cant find the right config to run the strip.

Maybe we need to solder a wire from DAT Pin to ESP?

golddragon007, 2020-06-06:
Hello,

I'm afraid that you can't run this without hardware modification because there's an ARM processor which fed by the ESP module. (https://github.com/Aircoookie/WLED/issues/398#issuecomment-597865330)

RomeyRome206, 2020-06-06:
[image]

Remove these 2 resistors, or cut their leads & you’ll be able to flash it. Then run a jump wire directly to the led strip from whatever pin you use for data.

golddragon007, 2020-06-06:
@RomeyRome206 hah, so I need to remove those two. Maybe I can do it. And then I connect directly the green wire with one of the ESP pins, right?
But the other guy why cut the first two pins leads of the ARM?

RomeyRome206, 2020-06-06:
> But the other guy why cut the first two pins leads of the ARM?

It effectively does the same thing - Disables ARM communication.  ESP and ARM rx/tx are linked together with resistors probably doing pull up/down duties. I’m guessing when you try to flash the ESP, the commands probably go to the ARM and one, or both reply and fail. You don’t even have to Remove the resistors. Just run a razor across their leads going to the arm.

golddragon007, 2020-07-02:
Removing the resistors did the trick, I was able to flash the boards after that.