arendst/Tasmota #5089: file (.h) not found will not upload

lincronin, 2019-02-01:
Sketch uses 582272 bytes (60%) of program storage space. Maximum is 958448 bytes.
Global variables use 50524 bytes (61%) of dynamic memory, leaving 31396 bytes for local variables. Maximum is 81920 bytes.
Uploading 586416 bytes from C:\Users\Linda\AppData\Local\Temp\arduino_build_724539/sonoff.ino.bin to flash at 0x00000000
................................................................................ [ 13% ]
................................................................................ [ 27% ]
................................................................................ [ 41% ]
................................................................................ [ 55% ]
................................................................................ [ 69% ]
................................................................................ [ 83% ]
................................................................................ [ 97% ]
.............                                                                    [ 100% ]
Invalid library found in C:\Program Files (x86)\arduino\libraries\lib: no headers files (.h) found in C:\Program Files (x86)\arduino\libraries\lib
Invalid library found in C:\Program Files (x86)\arduino\libraries\lib: no headers files (.h) found in C:\Program Files (x86)\arduino\libraries\lib**IMPORTANT NOTICE**
Is there a simple fix to this please if anyone knows be great?
I am quite new to all this so bear with me!

Thanks LInda

If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-02-01:
Hello Linda,
your setup is faulty. Try a portable install just for Tasmota. Described in wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

ascillato2, 2019-02-02:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-02 by ascillato2

lincronin, 2019-02-04:
i have , read the wiki, ensured i followed it to best of my knowledge, but it is still comingback to the above error only worse now that I have troubleshooted it.

Jason2866, 2019-02-04:
Have you downloaded Tasmota via green github download button?
Have you done a extra NEW portable Install of Arduino only for Tasmota?
Have you really all steps done as described in wiki?
Try it again 😀 You have missed something somewhere

ascillato, 2019-02-04:
@lincronin 

Please, try the chat for support, There are a bunch of great people there for helping you on that.