arendst/Tasmota #5090: I've created a new module for Tuya plug with power monitor working

jnalley20, 2019-02-01:
For Tuya plug:  https://www.amazon.com/gp/product/B07B9XCY5M/ref=ppx_yo_dt_b_asin_title_o00__o00_s00?ie=UTF8&psc=1

  { "Tuya Smart Plug1",       // Tuya S01 or AWP04L with Energy Monitoring
  //https://www.amazon.com/gp/product/B07B9XCY5M/ref=ppx_yo_dt_b_asin_title_o00__o00_s00?ie=UTF8&psc=1
     GPIO_LED2_INV,    // GPIO00 Red Led (1 = On, 0 = Off)
     0,     0,
     GPIO_NRG_SEL_INV,     // GPI03  BL0937 or HJL-01 Sel output (0 = Voltage)
     0,                // GPIO04
     GPIO_HJL_CF,      // GPIO05 BL0937 or HJL-01 CF power
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_REL1,        // GPIO12 Relay (0 = Off, 1 = On)
     GPIO_KEY1,        // GPIO13 Button
     GPIO_NRG_CF1,     // GPIO14 BL0937 or HJL-01 CF1 current / voltage
     GPIO_LED1_INV,    // GPIO15 LED Blue
     0, 0
  }

Jason2866, 2019-02-01:
You can add it to wiki :-)

ascillato2, 2019-02-02:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-02 by ascillato2

erwill2, 2019-02-03:
@jnalley20 How do we setup energy monitoring is it on compile or can you use generic? I'm able to set everything else just don't see anything for BL0937 or HJL-01.  Any Help would be greatly appreciated. Thanks

meingraham, 2019-02-03:
You can flash your Tuya device with Tuya-Convert and once it has TASMOTA on it, you can OTA flash it with a different version of pre-compiled TASMOTA firmware binaries available on thehackbox.org/tasmota

MrPasty, 2019-02-08:
I know this is closed but would also like to know the answer to the above, i could only get it to compile by removing the line with 0,0,0,0,0,0 in it however on flashing and selecting the Tuya module it hung and reset to basic sonoff

erwill2, 2019-02-09:
Here is what worked for me. Update with code below or compare with attached. 

@MrPasty

[sonoff_template.h.txt](https://github.com/arendst/Sonoff-Tasmota/files/2848099/sonoff_template.h.txt)

`{ "Tuya Smart1",       // Tuya S01 or AWP04L with Energy Monitoring

  //https://www.amazon.com/gp/product/B07B9XCY5M/ref=ppx_yo_dt_b_asin_title_o00__o00_s00?ie=UTF8&psc=1

  GPIO_LED2_INV,    // GPIO00 Red Led (1 = On, 0 = Off)
  GPIO_USER,
  GPIO_USER,        // GPIO02
  GPIO_NRG_SEL_INV, // GPI03  BL0937 or HJL-01 Sel output (0 = Voltage)
  GPIO_USER,        // GPIO04
  GPIO_HJL_CF,      // GPIO05 BL0937 or HJL-01 CF power
                             // GPIO06 (SD_CLK   Flash)
                           // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                    // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
  0,                // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
  0,                // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                    // GPIO11 (SD_CMD   Flash)
  GPIO_REL1,        // GPIO12 Relay (0 = Off, 1 = On)
  GPIO_KEY1,        // GPIO13 Button
  GPIO_NRG_CF1,     // GPIO14 BL0937 or HJL-01 CF1 current / voltage
  GPIO_LED1_INV,    // GPIO15 LED Blue
  GPIO_USER
  }`

MrPasty, 2019-02-09:
Perfect thank you @erwill2

evaluationevaluation, 2019-02-25:
Hi,

how did you find out the right GPIO setting for your device?

I'm currently struggeling with a Slitinto Dual Plug device and cannot find the right combination of settings using the template section in the Tasmota-Web.
Tried several dozen combinations but simply too much to guess without a glue of knowlegde of the semantic behind the different parameters: HLW8012CF, HLWBLCF1, HLWBLSEL, HLWBLSELi and BL0937CF.

Can you give me some hints?

DavinKD, 2019-02-25:
The Slitinto plugs are the same as this one.

https://blakadder.github.io/templates/sp201.html

You need to flash with the latest development bin.

Put this in your OTA URL in the Upgrade Firmware Screen

http://thehackbox.org/tasmota/sonoff.bin

After that, go to configuration and configure other.  Paste the template in and select "activate".

evaluationevaluation, 2019-02-25:
thanks for the quick answer but didn't work.
I'm using Tasmota 6.4.1 from yesterday. Your configuration is quite different from what I found out so far.
yours: {"NAME":"SP-201","GPIO":[31,0,0,131,17,134,0,0,21,18,132,22,0],"FLAG":0,"BASE":45}
mine: {"NAME":"Slitinto","GPIO":[17,255,255,255,134,132,255,255,255,130,22,21,18],"FLAG":0,"BASE":18}
Buttons and Relays are working with mine but the metering not...

Just pasted yours to check and got 3 relay buttons and shortly after a crash of the device and had to type in again the WIFI configuration using the AP... 
(in my web template code 31 means relay 3i)

What do I do wrong?

evaluationevaluation, 2019-02-25:
I think I found the solution. Based on your suggestion I searched for similar devices and found out that "Digoo DG-SP202" is quite similar and already known by Tasmota.
So far this configuration seems to work.

Would like to document this for other users but unsure how to do this.
Should I create a complete new section for Slitinto NX-SP202 device or simply note somewhere that it is (probably) equal to Digoo DG-SP202?

meingraham, 2019-02-25:
@evaluationevaluation 

6.4.1 is the release version. To get these features, you need the latest development version 6.4.1.x located at the link @DavinKD provided.

Development - http://thehackbox.org/tasmota/
Release - http://thehackbox.org/tasmota/release

From each of these repository pages you can choose which Core version you need.

Mike

evaluationevaluation, 2019-02-25:
@meingraham 
I downloaded the latest sources yesterday and compiled myself. Don't know the x-part of the version.
But wonder why the configuration mentioned by @DavinKD is quite different to mine.
Looking into sonoff_template.h I come to the conclusion that the configuration of @DavinKD could never work as it creates a 3rd relay [Code 31 on GPIO0]... don't understand this.

Would like to post my configuration for Slitinto device in order to help others but if I put it into this repo
[https://blakadder.github.io/templates/sp201.html] I will probably do more confusion than help...

meingraham, 2019-02-25:
@evaluationevaluation 

If you are compiling yourself, you'll need to ensure that you grab the development branch source code. On the main page, on the left hand side is the dropdown for which branch to select - master or development. Select Development and then download the source using the button on the right hand side.

I'm not familiar with the device(s) you're using so I cannot be of assistance.

Yes, once you have the config, create a template and contribute it using the new site @blakadder  created. Thanks.

Mike

DavinKD, 2019-02-25:
The extra relay is for the blue light on the buttons. I prefer to control them separately instead of configuring them as an led since the red led is already connected to the buttons.

I'm not sure how there could be multiple configurations for a Sp202, but who knows.

meingraham, 2019-02-25:
I can't speak to why your SP202 may be different. All of these devices though are made by a ton of manufacturers and who knows what mods they each have made.

Configure the LEDs as relays of their own... if the circuitry is independent of the onboard relay and if it is connected to one of the GPIOs. That way you can control the LED by turn via `PowerX` ON/OFF of that "relay". The Button(s) need to be configured to their corresponding GPIO. Tasmota will detect `ButtonX` and and take the corresponding action on `RelayX` (as @blakadder mentioned). You can set the behavior of the button for single-press, double-press, long-press, etc. See the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches) for those settings. You can also use `ButtonX#State` in [Rules](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules) to perform custom actions.

evaluationevaluation, 2019-02-25:
@meingraham 
Thanks for this clarification about source code download but it seems that I downloaded the development code by chance. Still confused why the configuration @blakadder for Teckin and OxaOxe use #31 (third socket) although the devices to not have a third socket. IMHO the configuration does not fit or I do not understand the code at all.
{"NAME":"SP-201","GPIO":[**31**,0,0,131,17,134,0,0,21,18,132,22,0],"FLAG":0,"BASE":45}

Just created a wiki page for Slitinto in "other devices".
Thank you for your advice!

DavinKD, 2019-02-25:
> Still confused why the configuration @blakadder for Teckin and OxaOxe use #31 (third socket)

Again, I created that template and the third "socket" which is actually relay3 is used for the blue LED that surrounds the two buttons.  Again, I could have configured it as LED1 or LED2, but I didn't want it to turn on when there is already a different red light.

DavinKD, 2019-02-25:
Also, it appears yours in not a US plug, which mine is.  Sorry I made the assumption yours was the same, but now I see your picture and it is different.

evaluationevaluation, 2019-02-28:
thank you all for the useful information.
@DavinKD: to be honest I oversaw your first answer according to relay3 but this makes then sense.
So far my module is now working find and a suitable configuration is placed in the blakadder github and mentioned in wiki.
My current working configuration (including Relay3i for blue led) is as follows:
{"NAME":"Slitinto SP202","GPIO":[17,0,0,0,134,132,0,0,131,31,22,21,91],"FLAG":0,"BASE":64}
Important is that GPIO0 is Button and not Relay3 in my case. So the SP202 devices are not really all the same but only similar.

DavinKD, 2019-02-28:
Yeah, sorry about that.  When you posted I saw "Slitinto Dual Plug" and assumed the US version SP201.

epicurean70, 2019-03-01:
Does anyone have the configuration of this UK type Tuya power plug with energy monitoring? I could get it to work with Blitzwolf SHP but the LED lights do not work
https://shopee.sg/product/64129412/1694422470

AS137430, 2019-03-01:
@epicurean70 : https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices,
Did you try GPIO0=Led1i ?

AS137430, 2019-03-02:
@epicurean70 : Somebody did the hard work for you. [www.earth.li/~noodles/blog/2018/12/power-monitoring.html](url)

epicurean70, 2019-03-03:
The link appears to be down.
ONly GPIO1 and GPIO3 are avaialble under Blitzwolf SHP. Are you suggesting
I configure module as generic?

blakadder, 2019-03-03:
you use Blitzwolf SHP as a starting point then follow https://github.com/arendst/Sonoff-Tasmota/wiki/Templates to modify for your own device

glentakahashi, 2019-05-10:
After banging my head against a wall for a long time, I found out that this configuration is also what randomly works for my Teckin SP20. I must have a very old/random version of that switch, which must happen to reuse the same board of the device you linked. 

THANK YOU SO MUCH

Erdcgh, 2019-10-19:
Can some one please give me a link to the compiled bin file for the AWP04L? I cannot get it to compile on my end. Thank you

Jason2866, 2019-10-19:
http://thehackbox.org/tasmota/sonoff.bin

Erdcgh, 2019-10-19:
Thanks Jason2866 for the quick respnds. Now do I just enter in the Template that is listed for the AWP04L on the Tasmota website, or is there a different Template that I need to use for this setup? Or do I just go and change the Module parameters? If so what would these be?

I currently have it as a generic module type and it has GPIO12 setup as relay1. It works fine but no power measurements are being shown.

Thanks again, for your help is greatly appreciated.

Jason2866, 2019-10-19:
Read wiki how to use and activate the template https://github.com/arendst/Sonoff-Tasmota/wiki/Templates#importing-templates
`{"NAME":"AWP04L","GPIO":[57,255,255,131,255,134,0,0,21,17,132,56,255],"FLAG":0,"BASE":18}`

Erdcgh, 2019-10-19:
That did it Jason2866. Thanks, and once more, Life Is Good!