arendst/Tasmota #5091: SmartConfig not working with 6.4.1

eugentib, 2019-02-01:
I thing something is wrong with my_user_config.h because with the latest release version I can't use WiFi Smart Config (I use it without problems on versions 5.x.x)

- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : Bare ESP12F
- [x] _Development/Compiler/Upload tools used_ : PlatformIO
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [x] _You have tried latest release or development binaries?_ : It works with sonoff-clasic
- [x] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE 
00:03:01 RSL: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
00:03:01 CFG: Use defaults
00:03:03 APP: Restarting
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
00:00:00 Project sonoff Sonoff Version 6.4.1(sonoff)-2_4_2
00:00:00 UPL: Arduino OTA Enabled Port 8266
00:00:00 WIF: SmartConfig active for 3 minutes
00:00:19 CMD: STATUS 0
00:00:19 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:19 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://myserver/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:20","StartupUTC":"","Sleep":50,"BootCount":1,"SaveCount":3,"SaveAddress":"F9000"}}
00:00:19 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-02-01T23:27:02","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
00:00:19 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":180,"SetOption":["00008009","558180C0","00000000"]}}
00:00:19 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":31,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","47082774","240083A0","00020096","000000C0"]}}
00:00:19 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7190","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"18:FE:34:A1:5C:16","Webserver":2,"WifiConfig":1}}
00:00:19 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"myhost","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A15C16","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:19 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:19 1970","Local":"Thu Jan 01 00:00:19 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"06:11","Sunset":"14:31"}}
00:00:19 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:19"}}
00:00:19 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:19","Uptime":"0T00:00:20","Vcc":3.426,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":4,"POWER":"ON","Wifi":{"AP":1,"SSId":"","BSSId":"64:70:02:63:51:4A","Channel":1,"RSSI":100}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
my_user_config.h
```/*
  my_user_config.h - user specific configuration for Sonoff-Tasmota

  Copyright (C) 2018  Theo Arends

  This program is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#ifndef _MY_USER_CONFIG_H_
#define _MY_USER_CONFIG_H_

/*********************************************************************************************\
 * This file consists of TWO sections.
 *
 * SECTION 1:
 * The first section contains PARAMETERS overriding flash settings if define CFG_HOLDER is CHANGED.
 * All parameters can be persistent changed online using commands via MQTT, WebConsole or Serial.
 *   Corresponding MQTT/Serial/Console commands are shown in [brackets]
 *
 * SECTION 2:
[... 393 lines cut ...]
//#define USE_KNX_NO_EMULATION                     // Create sonoff-knx with KNX but without Emulation
//#define USE_DISPLAYS                             // Create sonoff-display with display drivers enabled
//#define BE_MINIMAL                               // Create sonoff-minimal as intermediate firmware for OTA-MAGIC

/*********************************************************************************************\
 * No user configurable items below
\*********************************************************************************************/

#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
#endif

#endif  // _MY_USER_CONFIG_H_
```

**Expected behavior**
I am unable to set the wifi credentials with 3 android apps that I tried and even that in the serial monitor appears "WIF: SmartConfig active for 3 minutes" I can see a wifi ap called "ESP_A15C16" but with no port open (looks more like WIFI_MANAGER that I can't use because of the TLS) 

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-02:
>I can see a wifi ap called "ESP_A15C16

Tasmota don't provide that name. So, that means that your firmware is corrupt. May be it has rests of other firmware in flash. 

Please do a full erase with esptool.py as explained in the wiki and then flash with esptool.py the precompiled bins. 

If you need further assistant please find us in the Tasmota Support Chat. Github issues are meant for bug reporting. Help and support is better and faster in the chat.
Thanks

ascillato2, 2019-02-02:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-02 by eugentib

ToRvaLDz, 2019-02-28:
I'm having the same issue, I erased many times and flashed again with esptools or platformio, I  also tried to change cfg holder without success. The WiFi ap starts with ESP_ with smart config and it doesn't work, while it is ok with wifi manager where AP Name starts with sonoff_. Wifi manager has another issue: if I set the admin password in config after enter the password it redirect me to a blank page so I cannot setup WiFi connection, no problem without setting the admin password.

pzgdiego, 2019-05-13:
@ascillato @eugentib I was able to resolve the issue by modifying [line 159 #ifdef of support_wifi.ino](https://github.com/arendst/Sonoff-Tasmota/blob/0b6c62f5f929fd621aeecfd2dd3109d88ed2e025/sonoff/support_wifi.ino#L159) as follows:
```
#ifdef USE_SMARTCONFIG
    else if (WIFI_SMARTCONFIG == wifi_config_type) {
      AddLog_P(LOG_LEVEL_INFO, S_LOG_WIFI, PSTR(D_WCFG_1_SMARTCONFIG " " D_ACTIVE_FOR_3_MINUTES));
      WiFi.mode(WIFI_STA);
      WiFi.beginSmartConfig();
    }
#endif // USE_SMARTCONFIG
```

As you can see, I just added 
`
WiFi.mode(WIFI_STA);`

eugentib, 2019-05-13:
Good job, thanks!