arendst/Tasmota #5094: Sonoff SV with Tasmota - Turns on and off when switching power ON

hybrixxx, 2019-02-02:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Development/Compiler/Upload tools used :
- [x ] Hardware used : 
- [x ] If a pre-compiled release or development binary was used, which one? :
- [x ] You have tried latest release or development binaries? :
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
I'm using sonoff SV as wifi relay for my garage door. Everything is fine except that when I plug it, for first time when it initiates it switchs on and off the relay, so when I unplug and plug it, the garage door opens... Is there any way to stop this behavior?

Apart from this, I have an issue with the webclient password... I put a password to access the tasmota web client but there wasn't any place to set the user, so now I'm unable to access to the web console because it asks for User and Password both... not sure if the User is generic or if it was in another tab that I didn't see... if there's a default user and you could share...

Thanks!

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-02:
Hi,

The on and off at the begining is a known hardware issue. At boot time the esp8266 flips the state of some pins and it is not controlled by the software. You can try connecting your relay to another gpio. 

The username is set at compilation time only. The default that came with the precompiled bins is admin.

cbugeja, 2019-02-02:
@hybrixxx  Make sure that the message to MQTT server does not have a retain flag. If the message to the MQTT server is sent with a retain flag, i.e. you tell the MQTT server to retain the message, it will result in the behaviour that you are describing, not only at startup but also when the wifi connection is reconnected (after a disconnection). First send a MQTT message "OFF" with a retain flag to your device, then change whatever software you use to interface with the MQTT server (eg openHAB, Home Assistant...) to not retain the "ON" message.

hybrixxx, 2019-02-04:
I'm afraid the fix is not working:

after clean payload:
cmnd/sonoff_sv_garage_fb/POWER (null)
cmnd/sonoff_sv_garage/POWER (null)
cmnd/garagestate/POWER2 (null)
cmnd/sonoff_sv_garage_fb/POWER1 (null)
tele/sonoff/LWT Online
cmnd/sonoff/POWER (null)

wait 10 secs.....unplug and plug:
tele/sonoff/INFO1 {"Module":"Sonoff
SV","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/sonoff_sv_garage_fb/","GroupTopic":"sonoffs"}
tele/sonoff/INFO2
{"WebServerMode":"Admin","Hostname":"sonoff_sv_garage","IPAddress":"192.168.1.191"}
tele/sonoff/INFO3 {"RestartReason":"Power on"}
stat/sonoff/RESULT {"POWER1":"OFF"}
stat/sonoff/POWER1 OFF
stat/sonoff/RESULT {"POWER2":"OFF"}
stat/sonoff/POWER2 OFF
stat/sonoff_sv_garage_fb/RESULT {"POWER1":"ON"}
stat/sonoff_sv_garage_fb/POWER1 ON
stat/sonoff/RESULT {"POWER1":"OFF"}
stat/sonoff/POWER1 OFF

As you can see, it automatically published the ON message, I think it's not
homeassistant neither mqtt the ones creating the issue.

Thanks

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

cbugeja, 2019-02-04:
@hybrixxx 
Can you use MQTT box or a similar program to inject an OFF message with a retain flag to stat/sonoff_sv_garage_fb/POWER1

That should help.

ClayF, 2019-02-12:
Was facing this same issue.

Ensure that ```SwitchRetain 0``` and ```PowerRetain 0```.

What fixed it was changing the MQTT topic

ssinseeme, 2019-04-27:
I tried all the above and not able to resolve the problem.
```
sudo service mosquitto stop
sudo rm /var/lib/mosquitto/mosquitto.db
sudo service mosquitto start

I dont understand this part, where should i do this.
if you have an easier solution please share
```

ssinseeme, 2019-04-27:
I got it!!!!
I changed the topic but not from console that doesn't work. change topic in Sonoff MQTT config.