arendst/Tasmota #5095: TM1638 Json string output with Status 8

Jason2866, 2019-02-02:
**Describe the bug**
Output for TM 1638 with Status 8 is:

```
16:51:49 CMD: status 8
16:51:49 MQT: stat/sonoff/STATUS8 = {"StatusSNS":{"Time":"2019-02-02T16:51:49","Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"ON"}}
```
Should be (to be like the others):
```

`{"StatusSNS":{"Time":"2019-02-02T16:51:49,""TM1638":{"Switch1":"ON","Switch2":"ON","Switch3":"ON","Switch4":"ON","Switch5":"ON","Switch6":"ON","Switch7":"ON","Switch8":"ON"}}}
`
```

arendst, 2019-02-02:
Will change

Jason2866, 2019-02-02:
Ahh!

closed 2019-02-02 by Jason2866

jziolkowski, 2019-02-06:
But then, why does MPR121A return them nested? I get the switch/button difference, but shouldn't it be consistent?

 "MPR121A":{"Button0":0,"Button1":0,"Button2":0,"Button3":0,"Button4":0,"Button5":0,"Button6":0,"Button7":0,"Button8":0,"Button9":0,"Button10":0,"Button11":0,"Button12":0},"TempUnit":"C"}}

arendst, 2019-02-06:
@jziolkowski those buttons have no relation with the default buttons.

The switches of tm1638 do have a direct relation to the default buttons. The switch json is legacy style.