arendst/Tasmota #5100: MQTT keepalive interval

iomax, 2019-02-03:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
There seems to be no option to customize the MQTT keepalive interval ( fixed to 10 sec. )

**Describe the solution you'd like**
the option to personalize the MQTT keepalive interval

**Describe alternatives you've considered**
fix the new value in the code at the compile time.

**Additional context**
_Add any other context or screenshots about the feature request here._

Please evaluate to add the option to configure the MQTT keepalive interval today fixed at 10sec. Some times a value of 30 or even 60 can be acceptable and in the environments with lot of the devices it can help to lower the basic network traffic as mqtt server log as well.

Thanks

andrethomas, 2019-02-03:
MQTT_KEEPALIVE is defined in PubSubClient.h which may be found in the PubSubClient-EspEasy-2.6.09 folder under libraries.

If you have a special use case for a different value it is recommended to change it there before compiling your custom use case firmware/binary.

andrethomas2, 2019-02-03:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-03 by andrethomas2

iomax, 2019-02-03:
I agree on that and you can change at the code level but I think binary compilation for newbies, like me, can fail due to many reasons so I'd like to suggest it as an standard configurable option.

iomax, 2019-02-03:
indeed one tasmota device on my setup is flooding the mqtt server log. Maybe due to something more complex at mqtt client and/or server level or just a poor wifi coverage case, I don't know and I must investigate deeply, any way log filled at 10 seconds messages rate didn't help for sure. 

Anyway, thanks for the interest and the prompt feedback.

iomax, 2019-02-03:
6.4.1.10(eab6be8-basic) bin and 2.5.0beta2 core for all devices around my house. 
Indeed, checked it one more time and discovered it as 2.4.2 (!!!). Time to reflash it :-)

Thanks for the hint.

ayavilevich, 2020-11-04:
Related
https://github.com/arendst/Tasmota/issues/5341#issuecomment-721891563