arendst/Tasmota #5103: Connect to hidden SSID? Possible?

KamuKaroly, 2019-02-03:
Dear All,

I have a hidden SSID network.
Is it possible to connect the sonoff to this?
If so, how?

Thanks,
Gabor

nibblerrick, 2019-02-03:
Just put in your SSID and  WPA-Password.
Works flawless on my hidden SSID, no special configuration.

KamuKaroly, 2019-02-03:
I tried, but unfortunately it didn't work.
I'll try again.
Thanks!

andrethomas, 2019-02-03:
Just curious... why do you hide your SSID?

nibblerrick, 2019-02-03:
For me it's at the moment more a cosmetic reason. With the APs I have I can have multiple SSIDs and therefore I made an extra SSID for the IOT/M2M-stuff, so it is a bit seperated from the main Wificlients.
Lateron (when there is time(TM)) I like to put them in a seperate VLAN. At the moment it may be not so easy to sniff wifitraffic if you are in the main wifinetwork, but that's all. But that's more a reason for a second SSID. Hidden it is mainly to have less confusion when others at home search for wifi. And the IOT-Wifi it not sooooo obvious to see. But that's only a very little reason as we all know how easy it is to see a hidden SSID.

andrethomas, 2019-02-03:
Right, was just curious.

Please check that your SSID and password does not contain any escape characters, especially \ and % and @ 

I just tested and can connect to SSID which is not being broadcast (i.e. hidden SSID)