arendst/Tasmota #5104: Sonoff bridge question.

PMBonne, 2019-02-03:
I try to figure out how the bridge works. If I program the bridge to send a code like 584267, then I receive a code that are 08EA4B.. anyone that know what kind of code that are? Or how I can decide my own specific code?

ascillato2, 2019-02-04:
Yes, you can receive and send simple codes in 433 mhz. 

If you need further information, please check the wiki at sonoff bridge. 
If you need further assistant please find us in the Tasmota Support Chat. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-04 by ascillato2

Jason2866, 2019-02-04:
08EA4B (hex) = 584267 (dec)