arendst/Tasmota #5106: Can't have one FW that supports  S31 (and Pow2) and Displays

Frogmore42, 2019-02-04:
Yes, it does not make sense to have display support on an S31, since that would be unsafe for any usage.  But, I like to use the same FW image on multiple/all of my devices.  I have some that have displays and the S31s do not.  With the current setup, it is impossible to configure one FW image to meet all my needs.

**Describe the solution you'd like**
A configurable option to allow having Displays and Energy Monitoring

**Describe alternatives you've considered**
the other option is to create one image for energy monitoring and another for everything else, which is workable.

Better documentation on this space saving choice would be good.  It took me a couple of hours to figure it out.

arendst, 2019-02-04:
Be my guest and roll your own. As you already noticed it's all about memory usage, flash usage and equal I2C addresses. In my_user_config.h you find some figures which may give an indication for your choices.

I've made the current choices based on general use and hopefully not to many issues raised due to lack of memory (See TLS related issues related to memory issues too).

My local user_config_override.h contains a lot of disabled features but I still need at least two versions to satisfy the display drivers while I want them to be below 500k (which I prefer).

Frogmore42, 2019-02-04:
The problem is sonoff_post.h.  When you select USE_SENSORS and USE_DISPLAYS, the latter wins, and disables all the sensors.  Probably a simple check for USE_SENSORS before turning off stuff might be good, but I don't know your intention.

I definitely agree the default shouldn't have everything and making two flavors of the FW is not too hard, but PlatformIO doesn't really make it easy.