arendst/Tasmota #5107: pulseTime() edit

elgranshaman, 2019-02-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [] Development/Compiler/Upload tools used :
- [] Hardware used : 
- [] If a pre-compiled release or development binary was used, which one? :
- [] You have tried latest release or development binaries? :
- [] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**
Hello,

How can edit the source files of the function pulseTime()? I would like to make it so the relay is only closed for 25ms. Which might not be possible but because of HW limitations (too fast for a relay?). Anyhow, where do I find the file responsible for controlling this?

Thank you

localhost61, 2019-02-04:
As minimal loop is 100ms, changing the pulseTime duration won't help. You'll have to hard code in another way to achieve your goal.

ascillato2, 2019-02-04:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-04 by ascillato2

elgranshaman, 2019-02-05:
Why do you close this thread? My question has not been answered yet

elgranshaman, 2019-02-05:
I am trying to find the file where the variables of the function pulseTime() are defined

ascillato, 2019-02-05:
25ms for a relay is too fast. It is close to the time it takes to switch, so may be it can't even make contact.

Anyway, Tasmota is not prepared for that fast switching. A lot of changes should be done to support that. 

I think it is way better to use an arduino and a simple sketch to do that.

localhost61, 2019-02-05:
@ascillato 
For the relay of a Sonoff Basic it seems doable, mine uses a [FANHAR-W11-1A2STE](https://www.alibaba.com/product-detail/Hot-selling-FANHAR-W11-1A2STE-5VDC_60553617588.html) and Operate/Release time is about 5ms. But I agree that it's border line to try to do that within a so big project and with so little knowledge of how everything works together.

elgranshaman, 2019-02-06:
Thanks for your comments guys.

I agree that this requirement might be beyond the capabilities of the current HW. However, I would like to experiment and see what happens.

If I were to edit the source code that handles this request. Where do I find such files? Are these not visible on the available package to download?

I just would like to see if I can address this problem purely by SW. I would need to edit the ISR for the ESP8266 the Sonoff has right?

Thanks again for the help

localhost61, 2019-02-06:
Good luck !