arendst/Tasmota #5108: Add support for MLX90614 non-contact infrared thermometer

inforsgit, 2019-02-04:
I've looked through the documentation and searched around but there don't seem to be any references to potential support for the MLX90614 non-contact infrared thermometer in Tasmota. It would be very useful to have this included so it could be hooked up to modules running Tasmota.

The device is a non-contact infrared thermometer (https://www.melexis.com/en/product/MLX90614/Digital-Plug-Play-Infrared-Thermometer-TO-Can) that connects via I2C using address 0x5a and gives two values - the ambient temperature and the temperature derived from the non-contact IR sensor which can be used to measure the temperature of a remote object.

SparkFun have an example page for hooking up (and code) for the device here: https://learn.sparkfun.com/tutorials/mlx90614-ir-thermometer-hookup-guide and it is already integrated into ESPEasy.

mike2nl, 2019-02-05:
@inforsgit 
then you got the first issue. The addess 0x5a is already used by the CCS811 sesnor.
On the other site i have that sesnor already now for 2 month but no time to add the code to tasmota:

https://nl.aliexpress.com/item/CJMCU-MLX90614ESF-DCI-Infrarood-Temperatuur-Sensor-IIC-Communicatie-Module-MLX90614-Hoek-Precisie-IR-Meting/32846740665.html?spm=a2g0s.9042311.0.0.6ce34c4d7lgFZl

[image]

[image]

gemu2015, 2019-02-05:
@mike2nl 
i have a preliminary support for MLX90615. it has the address 0x5b and is very similar to MLX90614.
its on my fork. you may give it a try.

mike2nl, 2019-02-05:
@gemu2015 
thanks for the info, super.
Will look at it, possible change some little things and it will work.

sapkra, 2019-05-06:
@mike2nl  Any news? Is it working for you?

ascillato2, 2019-06-15:
Closing this issue as a Pull request has been submitted for this by @gemu2015

Thanks everyone for working on this. :+1:

closed 2019-06-15 by ascillato2