arendst/Tasmota #5109: MQTT connection keeps rebooting

max-rh, 2019-02-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
Hello there,

I have a sonoff bridge flashed with tasmota, it keeps connecting to the MQTT broker as seen in the console:

23:49:20 MQT: Attempting connection...
23:49:20 MQT: Connected
23:49:20 MQT: home/rf/LWT = Online (retained)
23:49:20 MQT: home/rf/cmnd/POWER = 
23:51:35 MQT: Attempting connection...
23:51:35 MQT: Connected
23:51:35 MQT: home/rf/LWT = Online (retained)
23:51:35 MQT: home/rf/cmnd/POWER = 

I am seeing this issue on my other tasmota sonoff devices as well, i tried the beta releases with no luck too, i switched to another mqtt broker but it didnt solve the issue.

Here is the info from my device:
Program Version	6.4.1(sonoff)
Build Date & Time	2018-12-24T14:41:00
Core/SDK Version	2_4_2/2.2.1(cfd48f3)
Uptime	0T00:26:17
Flash write Count	35 at 0xF9000
Boot Count	12

i have turned on all debug tools and noticed that the device keeps checking the wifi signal every 20 sec, is that normal?
14:56:54 WIF: Checking connection...
14:56:54 WIF: Connected
14:57:14 WIF: Checking connection...
14:57:14 WIF: Connected
14:57:34 WIF: Checking connection...
14:57:34 WIF: Connected

Any Advice ?

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ x] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : itead sonoff rf bridge and other devices from sonoff
- [ x] _Development/Compiler/Upload tools used_ : esptools
- [ x] _If a pre-compiled release or development binary was used, which one?_ : release 6.4.1
- [x] _You have tried latest release or development binaries?_ : yes
- [x] _Provide the output of command_``status 0`` :
```
15:04:19 MQT: home/rf/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"rf","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
15:04:19 MQT: home/rf/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:30:59","StartupUTC":"2019-02-04T13:33:20","Sleep":50,"BootCount":12,"SaveCount":35,"SaveAddress":"F9000"}}
15:04:19 MQT: home/rf/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
15:04:19 MQT: home/rf/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["Linksys01706",""],"TelePeriod":300,"SetOption":["00008029","558180C0","00000000"]}}
15:04:19 MQT: home/rf/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
15:04:19 MQT: home/rf/STATUS5 = {"StatusNET":{"Hostname":"Sonoff_RF_Bridge","IPAddress":"192.168.2.63","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:CB:3D:50","Webserver":2,"WifiConfig":4}}
15:04:19 MQT: home/rf/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CB3D50","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:04:19 MQT: home/rf/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 04 14:04:19 2019","Local":"Mon Feb 04 15:04:19 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:16","Sunset":"17:51"}}
15:04:19 MQT: home/rf/STATUS10 = {"StatusSNS":{"Time":"2019-02-04T15:04:19"}}
15:04:19 MQT: home/rf/STATUS11 = {"StatusSTS":{"Time":"2019-02-04T15:04:19","Uptime":"0T00:30:59","Vcc":3.462,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Linksys01706","BSSId":"58:EF:68:B7:13:9D","Channel":1,"RSSI":44}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
The issue keeps happening every few seconds

**Expected behavior**
the device keeps disconnecting and connecting to the mqtt broker

**Screenshots**
_If applicable, add screenshots to help explain your problem._
[image]

logs from my broker

Jason2866, 2019-02-04:
You can try http://thehackbox.org/tasmota/release/020300/sonoff.bin
See here for explanation https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

max-rh, 2019-02-04:
Thanks alot @Jason2866 , I flashed 020500 and it has been running for 30 min without any issues,
do you recommend that i update once this get of of the BETA version via OTA, or do u recommend always flashing the new firmwares ?

Jason2866, 2019-02-04:
If you flash via OTA and you have no issues all fine. If you encounter strange problem (s)
erasing whole flash (esptool erase_flash) and flashing a precompiled Tasmota is a good idea...
I have many of my devices running with 2.5.0 beta3 and encountered no problems
seems quit stable.... But still beta (but IMHO already better than all cores before)

max-rh, 2019-02-04:
Will do. Thanks again for your help.
I am kinda new to tasmota, if may i ask; what do u mean by IMHO ?