arendst/Tasmota #5113: Possible bug in Alexa code

scargill, 2019-02-04:
As you likely know I am the main author of ESP-GO - so I'm familiar with coding ESPs and Alexa. But I prefer Tasmota for Sonoffs....

You did such a good job with Tasmota and Sonoffs I decided not to reinvent the wheel... SO I have several Sonoff BASICS running the latest Tsamota from the sonoff-basic rom.

I'm only doing on-off, no sensors... and when I disconnect a sonoff (pulling the actual plug out) from the mains and reconnect, sometimes Amazon is griping it can't find the device - but messages rfom Node-Red via MQTT are fine (cmnd/power-five/POWER:ON) for example. 

To test this - for the first time I just added node-red-contrib-amazon-alexa to my Pi... and after Alexa failed on POWER-FIVE I added to amazon-alexa a non-existing device - toilet-light - and had that send on-off commands to POWER-FIVE by changing the topic to "cmnd/power-five/POWER") - talking to Amazon "computer - turn toilet light on" - works - "computer turn power five on"
 still says "power-five isn't responding - please check its networkr connection and power supply".... which can ONLY be th SONOFF and theTasmota Alexa code - the rest of the code has to be working or the MQTT from NR would not get through and turn on/off the device? Corruption?

I had a Sonoff go down yesterday - same scenario, plug out and back in - this time - totally dead - but thats not normal.

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _If a pre-compiled release or development binary was used, which one?_ :
- [ ] _You have tried latest release or development binaries?_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-02-04:
I doesnt have this behaviour. If i unplug, plug it, Alexa works without a issue.
Please provide Status 0 output from a device that fails.

scargill, 2019-02-04:
ok, you;ve lost me - Status 0 output - I have status/power-five/POWER - thats the only one I'vev used so far. I can check whatever you like on that power-five board right now if you're around, it looks like the AMAZON bit is gone for good (or at least until a reflsah) as I've cycled yte power on it again. Control via MQTT fine, control via node-red-amazon-echo fine, but same result talking to power-five... not responding.

Jason2866, 2019-02-04:
Take a look here
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management
Search for command `Status 0`
We need the infos all around your version you use....

scargill, 2019-02-04:
HOLD THE PHONE... I took the power off power-five, and reconnected and TIS time all is well, so it's not necessarily permanent. The board is now responding to NR MQTT, NR Amazon and Tasmota Amazon perrfectly. Why would thast have happened...... I wonder.. and what was that status command you asked about? Question. To get status I sent via mqtt topic.... cmnd/power-five/Status  and topic 0    - I've no idea how to erad that back.... topics stat/power-five, stat/power-five/Status stat/power-five/POWER give no output after that inject. Me being thick I guess?

Jason2866, 2019-02-04:
Easiest way. Open your webrowser and type in the ip of this device.
You have a webconsole in Tasmota. Just type here `Status 0`
Copy & Paste the output

scargill, 2019-02-04:
Indeed - easy. Hope it means more to you than me in terms of finding anything wrong. It's been  fine since the Amazon part came back to life. 

23:28:35 MQT: stat/power-five/POWER = ON
23:29:15 MQT: stat/power-five/RESULT = {"POWER":"OFF"}
23:29:15 MQT: stat/power-five/POWER = OFF
23:31:16 CMD: Status 0
23:31:16 MQT: stat/power-five/STATUS = {"Status":{"Module":1,"FriendlyName":["power five"],"Topic":"power-five","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:31:16 MQT: stat/power-five/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T05:36:14","StartupUTC":"2019-02-04T16:55:02","Sleep":50,"BootCount":53,"SaveCount":213,"SaveAddress":"F5000"}}
23:31:16 MQT: stat/power-five/STATUS2 = {"StatusFWR":{"Version":"6.4.1(basic)","BuildDateTime":"2018-12-24T14:43:30","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
23:31:16 MQT: stat/power-five/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["office","bedroom"],"TelePeriod":300,"SetOption":["00008009","55A180C0","00000000"]}}
23:31:16 MQT: stat/power-five/STATUS4 = {"StatusMEM":{"ProgramSize":445,"Free":556,"Heap":20,"ProgramFlashSize":4096,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":2,"Features":["00000809","0F082780","000083A0","21021706","000001C0"]}}
23:31:16 MQT: stat/power-five/STATUS5 = {"StatusNET":{"Hostname":"power-five","IPAddress":"192.168.14.118","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.14.1","Mac":"5C:CF:7F:81:2C:32","Webserver":2,"WifiConfig":2}}
23:31:16 MQT: stat/power-five/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.14.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_812C32","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:31:16 MQT: stat/power-five/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 04 22:31:16 2019","Local":"Mon Feb 04 23:31:16 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:16","Sunset":"17:51"}}
23:31:16 MQT: stat/power-five/STATUS10 = {"StatusSNS":{"Time":"2019-02-04T23:31:16"}}
23:31:16 MQT: stat/power-five/STATUS11 = {"StatusSTS":{"Time":"2019-02-04T23:31:16","Uptime":"0T05:36:14","Vcc":3.528,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"office","BSSId":"60:03:47:2D:90:43","Channel":6,"RSSI":100}}}

Alestrix, 2019-02-04:
I had a (maybe?) similar behavior today. I forgot why I did it, but I deleted my Hue-Emulated Sonoff Device from the device list in the Alexa app. After re-discovery the switch was suddenly discovered as a colored light bulb. Changing colors caused a "does not respond" error.
After deactivating hue emulation in Tasmota, deletion of the device in the Alexa app, re-discovery (which of course does not find anything), switching emulation on and again re-discovery in the Alexa app I again have a colored light bulb. This time changing colors causes the device to switch on, switching off is also possible. So it works, but shows strangely in the app.

My firmware is the same for a very long time now (5.13.1 - never change a running system) so no change there. It seems that Amazon has changed something in their discovery process very recently (was not like this a few days ago when I did something similar).

```
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Gartenlicht"],"Topic":"sonoff/gartenlicht","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.4.70/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:57","StartupUTC":"2019-02-04T22:06:46","Sleep":0,"BootCount":66,"SaveCount":223,"SaveAddress":"F5000"}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-06-14T00:01:40","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":4,"LogHost":"192.168.4.70","LogPort":514,"SSId":["rtlwrmft-IoT",""],"TelePeriod":300,"SetOption":["00008009","55C18000"]}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS4 = {"StatusMEM":{"ProgramSize":459,"Free":544,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff/gartenlicht-7339","IPAddress":"192.168.4.52","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"DC:4F:22:37:DC:AB","Webserver":2,"WifiConfig":2}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.4.70","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_37DCAB","MqttUser":"DVES_USER","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 04 22:08:43 2019","Local":"Mon Feb 04 23:08:43 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS10 = {"StatusSNS":{"Time":"2019-02-04T23:08:43"}}
23:08:43 MQT: sonoff/gartenlicht/stat/STATUS11 = {"StatusSTS":{"Time":"2019-02-04T23:08:43","Uptime":"0T00:01:57","Vcc":3.138,"POWER":"OFF","Wifi":{"AP":1,"SSId":"rtlwrmft-IoT","RSSI":66,"APMac":"92:2A:A8:17:94:33"}}}
```

scargill, 2019-02-04:
Well, my firmware is bang up to date - current as of at least this week. I'm betting that info is not helpful.

Alestrix, 2019-02-04:
If anything it shows that there was a change on Echo side.

scargill, 2019-02-04:
I got it out of Node-Red

04/02/2019, 22:52:20node: ef5b61f9.ed66a
stat/power-five/STATUS : msg.payload : string[275]
"{"Status":{"Module":1,"FriendlyName":["power five"],"Topic":"power-five","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}"
04/02/2019, 22:52:20node: ef5b61f9.ed66a
stat/power-five/STATUS1 : msg.payload : string[270]
"{"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"0T05:57:17","StartupUTC":"2019-02-04T16:55:02","Sleep":50,"BootCount":53,"SaveCount":215,"SaveAddress":"FB000"}}"
04/02/2019, 22:52:20node: ef5b61f9.ed66a
stat/power-five/STATUS2 : msg.payload : string[125]
"{"StatusFWR":{"Version":"6.4.1(basic)","BuildDateTime":"2018-12-24T14:43:30","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}"
04/02/2019, 22:52:20node: ef5b61f9.ed66a
stat/power-five/STATUS3 : msg.payload : string[170]
"{"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["office","bedroom"],"TelePeriod":300,"SetOption":["00008009","55A180C0","00000000"]}}"
04/02/2019, 22:52:20node: ef5b61f9.ed66a
stat/power-five/STATUS4 : msg.payload : string[200]
"{"StatusMEM":{"ProgramSize":445,"Free":556,"Heap":20,"ProgramFlashSize":4096,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":2,"Features":["00000809","0F082780","000083A0","21021706","000001C0"]}}"
04/02/2019, 22:52:20node: ef5b61f9.ed66a
stat/power-five/STATUS5 : msg.payload : string[204]
"{"StatusNET":{"Hostname":"power-five","IPAddress":"192.168.14.118","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.14.1","Mac":"5C:CF:7F:81:2C:32","Webserver":2,"WifiConfig":2}}"
04/02/2019, 22:52:20node: ef5b61f9.ed66a
stat/power-five/STATUS6 : msg.payload : string[184]
"{"StatusMQT":{"MqttHost":"192.168.14.71","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_812C32","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}"

scargill, 2019-02-04:
Yes but it was working at this point. It hasn't failed since. I don;t want to push my luck my pulling the power out ..

ascillato2, 2019-02-04:
Closing this issue as it is solved now. If you experience again this type of issues, please ask to reopen this issue. Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-04 by ascillato2