arendst/Tasmota #5114: Sonoff Basic wdt/excpetion reset

martin072, 2019-02-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE 1.8.7 (Mac)
- [x] Hardware used : Sonoff Basic ESP8266
- [x] If a pre-compiled release or development binary was used, which one? : Not used
- [x] You have tried latest release or development binaries? : 6.4.1.13
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```00:00:00 Project sonoff SONOFF Test Version 6.4.1.13(sonoff)-2_5_0_BETA3
00:00:00 WIF: Connecting to AP1 UPC873263 in mode 11N as sonoff-6690...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-6690 with IP address 192.168.2.161
00:00:05 RSL: INFO1 = {"Module":"Sonoff Basic","Version":"6.4.1.13(sonoff)","FallbackTopic":"cmnd/DVES_8ABA2222_fb/","GroupTopic":"sonoffs"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-6690","IPAddress":"192.168.2.100"}
00:00:05 RSL: INFO3 = {"RestartReason":"Hardware Watchdog"}
00:00:05 RSL: RESULT = {"POWER":"ON"}
00:00:05 RSL: POWER = ON
00:00:05 RUL: AM2301#TEMPERATURE>20.8 performs "power on"
00:00:05 RSL: RESULT = {"POWER":"ON"}
00:00:05 RSL: POWER = ON
00:00:13 RSL: STATE = {"Time":"1970-01-01T00:00:13","Uptime":"0T00:00:13","Vcc":3.499,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"UPC873263","BSSId":"70:4F:57:A6:AB:C3","Channel":5,"RSSI":62}}
00:00:13 RSL: SENSOR = {"Time":"1970-01-01T00:00:13","AM2301":{"Temperature":22.0,"Humidity":39.6},"TempUnit":"C"}
00:00:19 CMD: status 0
00:00:19 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["SONOFF Test"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:00:19 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.10:80/api/arduino/sonoff.ino.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:00:18","StartupUTC":"","Sleep":50,"BootCount":10,"SaveCount":51,"SaveAddress":"F9000"}}
00:00:19 RSL: STATUS2 = {"StatusFWR":{"Version":"6.4.1.13(sonoff)","BuildDateTime":"2019-02-01T22:27:09","Boot":6,"Core":"2_5_0_BETA3","SDK":"3.0.0-dev(c0f7b44)"}}
00:00:19 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":2,"LogHost":"192.168.2.10","LogPort":514,"SSId":["UPC873263",""],"TelePeriod":300,"SetOption":["00008001","558180C0","00000000"]}}
00:00:19 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":546,"Free":456,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","1FC8A794","000003A1","000000C6","000000C0"]}}
00:00:19 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-6690","IPAddress":"192.168.2.161","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"68:C6:3A:8A:BA:22","Webserver":2,"WifiConfig":5}}
00:00:19 RSL: STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:19 1970","Local":"Thu Jan 01 00:00:19 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":99,"Sunrise":"07:49","Sunset":"15:40"}}
00:00:19 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:19","AM2301":{"Temperature":22.0,"Humidity":39.5},"TempUnit":"C"}}
00:00:19 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:19","Uptime":"0T00:00:18","Vcc":3.499,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"UPC873263","BSSId":"70:4F:57:A6:AB:C3","Channel":5,"RSSI":64}}}

Irregular times the unit resets, sometimes it reports as last reset reason a wdt, sometimes an exception.

ascillato2, 2019-02-04:
Please, try to erase all flash with esptool.py as explained in the wiki and flash it again using the precompiled bins.

Jason2866, 2019-02-04:
You use a 2.5.0 beta. It can run stable.... you walk alone with beta versions No support

martin072, 2019-02-04:
Undestood. Just as a heads up for you guys as developers.

closed 2019-02-04 by martin072