arendst/Tasmota #5115: Lockdown remote device?

bwmtx, 2019-02-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

closed 2019-02-04 by bwmtx

ascillato, 2019-02-04:
Hi,

Sorry, Your issue is empty. 

If you want information, please check the wiki.
If you have usage questions, please ask in the Tasmota Support Chat
If you have config problems, please ask in the Tasmota Support Chat
If you have a project you want to develop, please ask in the Tasmota Support Chat and then may be will be added to the wiki
If you think you hit a bug, please complete the bug template
If you have a weird issue, please complete the troubleshooting template
If you have a feature request, please complete the feature request template

Thanks.

bwmtx, 2019-02-04:
Adrian,

That was super quick!  This is my first time posting, please be patience
due to my lack of understanding about how the forum works.  Anyways, please
see #5116.  Thanks in advance for your help!

Bobby