arendst/Tasmota #5116: Lockdown remote tasmota devices?

bwmtx, 2019-02-04:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
- [x] Hardware used : 
- [ ] If a pre-compiled release or development binary was used, which one? :
- [ ] You have tried latest release or development binaries? :
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Hello everyone...

I'd like to know how to lock down remote Tasmota devices?  I've disabled the serial monitor and web server.  Is that sufficient to stop someone from compromising my device to get the fingerprint, mqtt username / password & host, and wifi ssid / password?

If someone has physical access to the device could they use something like this >>> https://www.hackster.io/rayburne/avr-firmware-duplicator-115d8f

Thanks in advance for any advice / recommendations.

ascillato, 2019-02-04:
You can add an admin password in the extra configuration menu

If you want even more security, you can disable the webserver (you need to compile by yourself this type of version) and also you can enable TLS encryption for MQTT communication

The standard Tasmota firmware is meant for a safe wifi network. It is not meant to be connected directly to an open port to internet.

ascillato2, 2019-02-04:
Closing this issue as it has been answered.

Anyway, if you need further assistance, please just ask here or in the Tasmota Support Chat. Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-04 by ascillato2

bwmtx, 2019-02-04:
> You can add an admin password in the extra configuration menu.  If you want even more security, you can disable the webserver (you need to compile by yourself this type of version)

The web server is turned off.  Does this prevent http commands?

> you can enable TLS encryption for MQTT communication

TLS is enabled 

> The standard Tasmota firmware is meant for a safe wifi network. It is not meant to be connected directly to an open port to internet.
The device is on a "safe" network.  I'm just concerned about someone gaining access to the device and "hacking" it.

> The Passwords, SSID, etc in the EEPROM inside the companion memory for the ESP8266 are not encrypted. So, yes. If someone can access the device physically and know how to read the memory and know where to search it, can obtain all your data.
There is no way to prevent this?  This is achievable even with serial interface turned off?

> But a great thing about Tasmota is that all your data is stored just locally inside the device. Tasmota don't use any cloud service like the original firmware to send your data and your usage statistics.
Agreed!  What I want to do is use Tasmota to monitor some remote equipment that then connects to my server.  Once installed the device can be considered out of my "physical control".  The device isn't nefarious or anything... but I don't want someone removing it then obtaining confidential stuff which could potentially compromise my server.

ascillato2, 2019-02-04:
> Once installed the device can be considered out of my "physical control"

Ok, that is the main point of concern. Any device with ESP8266 is very easy to copy its firmware if you have physical access to it. That's it how the MCU is.

In your use-case, you need another type of device more secure. For example the ESP32 has the option to not allow anyone to read its firmware, but Tasmota is not supported on that MCU. Sorry. Tasmota was built for esp8266 devices like iTead Sonoffs.

bwmtx, 2019-02-04:
> In your use-case, you need another type of device more secure. For example the ESP32 has the option to not allow anyone to read its firmware, but Tasmota is not supported on that MCU. Sorry. Tasmota was built for esp8266 devices like iTead Sonoffs.

Understood.  Thank for the help