arendst/Tasmota #5117: New Sonoff Basics not working

scargill, 2019-02-04:
I'm sure something similar has been discussed elsewhere but not using the programming tool I use.
I just got a batchof Sonoff BASICs , th ones with heavy wires instead of thick tracks.

I just flashed two on a PC and got no-where with Sonoff-basic.rom - so I checked. I'm using NodeMCU Flasher on Windows and I'd mistakenly left the advanced settings at 4MB - so I changed that to 1MB (320400 baud as usual for me, 40Mz flash speed as usual and DIO mode).

I've reflashed both boards but still no lights after flsahing and they ar not showing up as access points on my Android phone which they normally do.

Ideas?

ascillato, 2019-02-04:
Try erase all flash with esptool.py as explained in the wiki and then please follow the wiki steps for flashing the precompiled bins from release.

ascillato2, 2019-02-04:
Remember that for support it is available the Tasmota Support Chat.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

scargill, 2019-02-04:
#4515 didn't help and I use the NODEMCU Windows flasher. Am I right in saying these are 1MB?  I tried flashing a 1MB blank file then the sonoff-basic.rom file again, no better off.  I don;t know where in the WIKI it refers to reflashing the Sonoff BASIC , not marked as R2 but definitely genuine and ith 2 heavy wires instead of tracks... do I need to do any more than use a 1 meg blank file to wipe/resurrect them?

ascillato, 2019-02-04:
I don't know if your flasher software delete all the flash content. There are some parts that contains the SDK info and wifi info. If you flashed 4M to a 1M it will be corrupt information.

A bullet proof method is: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#esptool-usage

scargill, 2019-02-05:
That fixed it, ascillato - I just messed up the file using 4MB setting. I used esptool.exe rather than .py but otherwise the same. I'd forgotten all about esptool.exe

For others in this situation, I had memory size wrong in NodeMC U Flasher AND the programming mode set to DIO (my normal).

These boards need DOUT. I'm now using NODEMCU Flasher successfully having first fixed my original 2 boards with:

esptool.exe --port COM15 erase_flash
then
esptool.exe --port COM15 write_flash -fs 8m -fm dout 0x0 sonoff-basic.bin

scargill, 2019-02-05:
/looks like I spoke too soon. The sonoffs light up ok after flashing - I can connect the phone to them and set up SSID and passwords (pimarfy and secondary  SSIDs and passwords) and then save and reboot - but they're not appearing on my network - but I've done 10s of the previous sonoff basic boards without this problem. Is the sonoff-basic.rom file ok with the new boards?

ascillato2, 2019-02-05:
Yes, it is ok. No issue at all. Are you using spaces, dashes or special characters in the ssid or password? Those are not supported.

scargill, 2019-02-05:
Looks like I should have stuck with sonoff-basic.bin as sonoff.bin works a treat but OTA presents a message up complaining about minimal firmware. I have one of the boards on sonof-basic operating fully now and OTA is working thanks.

scargill, 2019-02-05:
I use a period in the password incidentally.

ascillato2, 2019-02-05:
Great!

Closing issue as it is solved. Thanks

closed 2019-02-05 by ascillato2