arendst/Tasmota #5119: Wifi Initialized every few seconds AP1 SSId (RSSI)	BSMT (92%)

Jimmyneu, 2019-02-05:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used : na
- [x ] Hardware used : sonoff basic
- [ x] If a pre-compiled release or development binary was used, which one? : 6.2  & 6.4.1
- [ x] You have tried latest release or development binaries? : yes
- [x ] Provide the output of command ``status 0`` 
```
:06:17:58 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
06:17:58 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:46","StartupUTC":"2019-02-05T05:07:12","Sleep":50,"BootCount":2157,"SaveCount":2458,"SaveAddress":"FA000"}}
06:17:58 RSL: STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
06:17:58 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BSMT",""],"TelePeriod":300,"SetOption":["00008021","558180C0","00000001"]}}
06:17:58 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
06:17:58 RSL: STATUS5 = {"StatusNET":{"Hostname":"","IPAddress":"172.16.0.77","Gateway":"172.16.0.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.0.1","Mac":"84:F3:EB:92:2A:70","Webserver":2,"WifiConfig":5}}
06:17:58 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 05 05:17:58 2019","Local":"Tue Feb 05 06:17:58 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:14","Sunset":"17:53"}}
06:17:58 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-02-05T06:17:58","Switch1":"OFF"}}
06:17:58 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-02-05T06:17:58","Uptime":"0T00:10:46","Vcc":3.448,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BSMT","BSSId":"28:C6:8E:6E:8B:B1","Channel":1,"RSSI":100}}}
```

Can someone let me know what is going on below? Thank you for your time.

00:00:00 Project sonoff Sonoff Version 6.4.1(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 BSMT in mode 11N as ...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on .local with IP address 172.16.0.77
00:00:06 RSL: INFO1 = {"Module":"Sonoff Basic","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/DVES_922A70_fb/","GroupTopic":"sonoffs"}
00:00:06 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"","IPAddress":"172.16.0.77"}
00:00:06 RSL: INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 RSL: RESULT = {"POWER":"OFF"}
00:00:06 RSL: POWER = OFF
06:07:27 RSL: STATE = {"Time":"2019-02-05T06:07:27","Uptime":"0T00:00:15","Vcc":3.447,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BSMT","BSSId":"28:C6:8E:6E:8B:B1","Channel":1,"RSSI":100}}
06:07:27 RSL: SENSOR = {"Time":"2019-02-05T06:07:27","Switch1":"OFF"}
06:07:38 DNS: Initialized
06:07:58 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:08:08 WIF: Connected
06:08:08 DNS: Initialized
06:08:27 DNS: Initialized
06:08:48 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:08:54 WIF: Connected
06:08:54 DNS: Initialized
06:09:14 DNS: Initialized
06:09:34 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:09:40 WIF: Connected
06:09:40 DNS: Initialized
06:10:00 DNS: Initialized
06:10:20 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:10:24 WIF: Connected
06:10:24 DNS: Initialized
06:10:44 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:10:48 WIF: Connected
06:10:48 DNS: Initialized
06:11:08 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:11:12 WIF: Connected
06:11:12 DNS: Initialized
06:11:32 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:11:36 WIF: Connected
06:11:36 DNS: Initialized
06:11:56 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:12:02 WIF: Connected
06:12:02 DNS: Initialized
06:12:21 DNS: Initialized
06:12:27 RSL: STATE = {"Time":"2019-02-05T06:12:27","Uptime":"0T00:05:15","Vcc":3.447,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BSMT","BSSId":"28:C6:8E:6E:8B:B1","Channel":1,"RSSI":100}}
06:12:27 RSL: SENSOR = {"Time":"2019-02-05T06:12:27","Switch1":"OFF"}
06:12:42 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:12:46 WIF: Connected
06:12:46 DNS: Initialized
06:13:06 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:13:10 WIF: Connected
06:13:10 DNS: Initialized
06:13:30 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:13:36 WIF: Connected
06:13:36 DNS: Initialized
06:13:56 DNS: Initialized
06:14:16 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:14:22 WIF: Connected
06:14:22 DNS: Initialized
06:14:42 DNS: Initialized
06:15:03 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:15:06 WIF: Connected
06:15:06 DNS: Initialized
06:15:26 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:15:32 WIF: Connected
06:15:32 DNS: Initialized
06:15:52 DNS: Initialized
06:16:12 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:16:22 WIF: Connect failed with AP timeout
06:16:24 WIF: Connected
06:16:24 DNS: Initialized
06:16:44 WIF: Connecting to AP1 BSMT in mode 11N as ...
06:16:50 WIF: Connected
06:16:50 DNS: Initialized
06:17:10 DNS: Initialized

Program Version | 6.4.1(sonoff)
-- | --
2018-12-24T14:41:00
2_4_2/2.2.1(cfd48f3)
0T00:08:01
2458 at 0xFA000
2157
Software/System restart
Sonoff
 
BSMT (92%)
.local
172.16.0.77
172.16.0.1
255.255.255.0
172.16.0.1
84:F3:EB:92:2A:70
 
Disabled
 
None
Enabled
Web Server
 
9579120
0x14405E
1024kB
1024kB
521kB
480kB
21kB

**(Please, remember to close the issue when the problem has been addressed)**

cbennett6008, 2019-02-05:
This is a known issue. Try the suggestions on this wiki page.   https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

It's different for different people, but for me, seems to be influenced by:
Multiple ubuquity unify access points on 'auto channel'
Wifi congestion

The 2.5.0 core Beta3 branch firmwares work better for me than the 2.4.1 based firmwares, but nothing seems more tolerant than 6.4.1 on the 2.3.0 core
Hope that helps!

ascillato2, 2019-02-05:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-05 by ascillato2

Jimmyneu, 2019-02-05:
After flashing it looked to be working fine. When I connect it to A/C, so off basic reboots every 60/90 seconds. Can I do anything about it or call it defective?