arendst/Tasmota #5120: Limit switches with extra checking

jpadie, 2019-02-05:
I am trying to create a solution to provide for a robust rainwater sump pump.  The pump would kick in when the top switch is closed and stop when the bottom switch is open.

But I also need to test for conditions where the pump has burned out (the top switch remains closed for too long) and for when either switch is 'cloggy' (they can get coated with mud which may cause them to remain closed for too long.   So I wish to build in timers and alerts.

The question is whether this is better done as a complex set of rules or creating a special sensor module?

If the latter then is there a tutorial at all that helps us understand how modules are constructed and what the hooks are?

ascillato, 2019-02-05:
>The question is whether this is better done as a complex set of rules or creating a special sensor module?

Rules should work just fine for that and are already implemented. No need to code anything. 

>If the latter then is there a tutorial at all that helps us understand how modules are constructed and what the hooks are?

See sensors API in the wiki.

ascillato2, 2019-02-05:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-05 by ascillato2

jpadie, 2019-02-05:
> Rules should work just fine for that and are already implemented. No need to code anything.

rules are quite limiting and difficult to create flexibility for multi-layered conditionals.  I've had a few goes at trying to create conditionality through either variables or turning on/off other rules.  One concern over the rule on/off method is that this is written to flash I believe and there is a danger that the saving of a state is interrupted by an outage; and of course that the memory locations can become unwritable.

ascillato, 2019-02-05:
>rules are quite limiting and difficult

So as you have an already made opinion about rules, you have already chosen writing a C code for your solution, right? 

>the memory locations can become unwritable.

The program is on the flash companion chip next to the esp8266 chip. That memory chip is the same for flash program and eeprom. So if the memory brokes, will do for all the chip. Yes, some part can be wear down if you overpass the 100.000 write cycles. So, using rules or C code will be the same for the memory chip writting usage.

>this could be written in five minutes in normal C code

Ok, so you know C code very good. Well, just do it.

jpadie, 2019-02-05:
i have an opinion born from experience that rules make it difficult to handle multi-layered conditionals.  I am willing to be persuaded otherwise, of course.  Hence the discourse.

I have not chosen to write C as a solution.  The choice will be made once I am confident, following this discourse, of whether rules are sensibly viable.   However I know that writing a stand-alone sketch to handle the switch events would be a few minutes work - i have done similar things hundreds of times.

Yes - coding in C is straightforward.  However I am not experienced in the hooks and methods necessary to code within the Tasmota ecosystem.  Hence the request for information.  Thank you - I will read through the API; that may resolve the quandary.

the issue was not rules vs c-code re the situation for the unwritable flash chip.  It was using rule on/off as a pseudo conditional that was the concern. if a bit has 100000 write cycle nominal life, and the state of a rule changes, say, once a minute then that there is a risk of failure in less than three months.  Whilst a state change is probably not likely every minute it remains a low number. Perhaps there is a more flexible way of creating multilayered conditionality within rules?

ascillato, 2019-02-05:
>if a bit has 100000 write cycle nominal life, and the state of a rule changes, say, once a minute then that there is a risk of failure in less than three months.

Sorry, you are mixing RAM with Flash. 

The state of a variable is managed in RAM. You can change them all the times you want and you will not have any issue. 
Flash is where the program and configuration is stored.

jpadie, 2019-02-05:
I don't think I am mixing it up at all. 
the state of whether a rule is enabled or disabled is written to flash, in my understanding.  Is this not the case?

ascillato, 2019-02-05:
An advise is that you make the solution you need with the method is easier for you. If C code, just try it. For the flash memory writing cycles point of view will be the same. Just one write when you upload the new modified firmware with your custom driver or just one write when you add the rules in the console.

ascillato, 2019-02-05:
>state of whether a rule is enabled or disabled is written to flash

Yes, that is configuration but you should just enable rules once and use it. 
You should not being enabling and disabling rules every minute