arendst/Tasmota #5121: Use sensor DS18x20 address as identifier

JotarOne, 2019-02-05:
Hi!

This is my first request, so please be kind :)
My request is to have a setting enabling using the address of a DS18x20 sensor in stead of "DS" + index.

My reason being that I have 3 DS18B20 sensors in three different rooms and I need to know from which rooms the temperatures are coming. 
I am using Home Assistant to monitor the values.
When all three are present then all is well and I get the sensors DS1, DS2 and DS3.
The problem arises when, for some reason, one of the sensors stops working (due to cable issues or whatever). Then I still get DS1 and DS2 but not necessarily containing the same sensors as before.
I have not found a way in Home Assistant to configure a sensor to pick a value based on the value of the address property, so I configure them to fetch values from DS1, DS2 etc.
If one sensor should fail I can no longer be certain that a (Home Assistant) senor value originates from a specific room.

The solution, as I see it, would be to replace DS1, DS2 etc with the sensor address to be able to read sensor information more easily and, hopefylly, more reliable.

I have solved it for myself by changing the code in my sonoff sketch as follows (Sonoff-Tasmota-6.4.1).
The legacy code is used because I have not been able to get sensor values with "USE_DS18x20".

In my_user_config.h (row 284):
_#define DS18x20_LEGACY_ADDR_AS_ID                // Optional, for use with legacy only. Sends the sensor address (instead of DS1, DS2 etc) in the JSON string._

In xsns_05_ds18x20_legacy.ino (row 189):
_#ifdef DS18x20_LEGACY_ADDR_AS_ID
        snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s%s\"%s\":{\"" D_JSON_TYPE "\":\"%s\",\"" D_JSON_ADDRESS "\":\"%s\",\"" D_JSON_TEMPERATURE "\":%s,\"IX\":%d}"),
          mqtt_data, stemp, Ds18x20Addresses(i).c_str(), ds18x20_types, Ds18x20Addresses(i).c_str(), temperature, i +1);
#else
        snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s%s\"DS%d\":{\"" D_JSON_TYPE "\":\"%s\",\"" D_JSON_ADDRESS "\":\"%s\",\"" D_JSON_TEMPERATURE "\":%s}"),
          mqtt_data, stemp, i +1, ds18x20_types, Ds18x20Addresses(i).c_str(), temperature);
#endif //USE_DS18x20_ADDR_AS_ID_        

This gives me what I need to solve my problem.
In addition I've added an "IX" property to quickly be able to se which is which in the web interface,
By looking at other posts, I am guessing that the issue would be the same with "USE_DS18x20".

I am requesting this feature so it might be helpful to others and so that I don't have to keep track of my own revisions :)

Hopefully I haven't missed an obvious way to solve this or another post requesting the same functionality.

localhost61, 2019-02-06:
FYI, using 6.4.1.14 and `#define USE_DS18x20` in `my_user_config.h`
I get this, for 5 DS18B20 and one 4k7 pullup on GPIO1 of a Dual R2 :
`tele/tasmota/Home/D13/spare/SENSOR = {"Time":"2019-02-06T19:10:02","Switch1":"OFF","Switch2":"OFF","DS18B20-1":{"Id":"000004A1BD20","Temperature":17.1},"DS18B20-2":{"Id":"000004C32226","Temperature":17.3},"DS18B20-3":{"Id":"000004C39B3D","Temperature":17.2},"DS18B20-4":{"Id":"000005A4E16F","Temperature":17.3},"DS18B20-5":{"Id":"000005A52682","Temperature":17.1},"TempUnit":"C"}`

ascillato2, 2019-02-06:
Hi, 

As @localhost61 has explained, the ID is already implemented for latest driver.

Thanks for sharing your ideas.

closed 2019-02-06 by ascillato2

JotarOne, 2019-02-06:
Hi,
It does not address the issue I was trying to resolve.
The way this json is created it still has an index based array/list of values.
To me, and it may be just me not being knowledgeable enough, it seems a bit convoluted to get the tempereture value for a sensor with a specific address.
At least I haven't figured out how to do it in Home Assistant.
I would appreciate if someone better than me could show me how I could get the temperature value of a sensor using the address.
@localhost61 - Could you share how you wired your sensors?
I can't get it to work. I'm using a 4k7 resistor between 3.3v and data (have tried pins 14 and 1 so far) and not using the parasite option (but I have tried that to)
Any help would be appreciated.

localhost61, 2019-02-07:
@JotarOne 
Yes, pullup is between 3.3V and Data pin.
I never tried parasitic option, this is an extract of my `my_user_config.h` setup:
```
//#define USE_DS18x20_LEGACY                       // Optional for more than one DS18x20 sensors with dynamic scan using library OneWire (+1k5 code)
#define USE_DS18x20                                // Optional for more than one DS18x20 sensors with id sort, single scan and read retry (+1k3 code)
//  #define W1_PARASITE_POWER                      // If using USE_DS18x20 then optimize for parasite powered sensors
//  #define DS18B20_INTERNAL_PULLUP	           // Use INPUT_PULLUP internal pullup resistors for single DS18B20
```

[image]

[image]

In my application, the cables length are certainly shorter than yours, it is intended to monitor an oil boiler.
For best results with lengthy lines or lot of sensors, the pullup should be located as close as possible of the controller side and can be as low as 2kΩ if necessary (I already made a test bench with 20 sensors and had to use a 2.2kΩ pullup). On the longer lines, a serial resistor of about 100Ω may be inserted on the Data line to reduce noise effects. 

For practical reason, I solder [XH2.54](https://fr.aliexpress.com/item/XH2-54-Connecteur-M-ne-pin-T-te-m-le-mat-riau-2-54mm-pitch-XH/32848821355.html) connectors on all my Sonoffs.  For this one, I've fitted an SMD resistor on the sensor connector attached to the Sonoff (so that I can easily exchange a faulty Sonoff or test with an other sensor).
[image]

Then, comes 2m of a 2X2AWG24 cable, then a little PCB where the cables of all 5 sensors are soldered pin to pin. 4 of them are about 30cm long, the last one that is 1m long. This one was made several years ago.
[image]

For a single sensor, I wire them like below, add hot glue on the pullup and sensor wires to avoid short circuits and protect the whole with a shrinking tube that extrude the exceeding glue.
[image]

JotarOne, 2019-02-07:
@localhost61 
Very thurough decription, thank you!
Then it seems like I have wired the same way, and I am going to experiment with resistor values and see if I can get som life out of some sensors on a bread board.
Thank you again for a very informative answer!