arendst/Tasmota #5124: Sonoff T1 R2 Boot Looping After flashing firmware

Bloodyagent, 2019-02-05:
Hi All
I am new to Sonoff and Tasmota.
I have been struggling to get my Sonoff T1 R2 US with Esp8285 flashed and connected to Wifi.
I have had no issues flashing a sonoff basic or a NodeMCU and connecting it to wireless.
My initial issue was the unit was not connecting to my wifi network. I have been through thread after thread trying to resolve and have tried the following with no luck.
Different firmware as far back as 5.14. Different core binaries including 2.3.0, 2.40, 2.4.2.
I have done a full flash erase using esptool and even tried to program using a bin file.
Using putty I can see the unit refused to connect to wireless, even with no specials characters in SSID and password.

My issue now is:

After multiple reflash after reflash I now cannot get the unit working at all, its in a boot loop, and no matter what firmware I load, fully erasing flash, i get the exact same error. I'm also changing the CFG_HOLDER value each time.
My Ardunio IDE settings are:

Board: ESP8285
1M no spiffs
v2 Higher Bandwidth
Erase Flash only sketch

Error is below, it just repeats continuously. Would appreciate some assistance. Thanks in advance.

ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Exception (0):
epc1=0x402684c0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000c depc=0x00000000

ctx: sys
sp: 3fffe780 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffe920:  00000000 00000000 00000000 00000031
3fffe930:  00000005 00000002 3fff0116 402684b8
3fffe940:  00000006 3fff00da 3fff00c8 3fff00d8
3fffe950:  0000001f 3fffe998 00000000 00000000
3fffe960:  00000040 0000002f ffffffff 0000004b
3fffe970:  000000d2 00000138 00000006 ffffffa6
3fffe980:  00000138 00000000 00000002 40006784
3fffe990:  00000001 3fff00c8 3d43474b 40263539
3fffe9a0:  00000000 4026898c 00000000 3fff00c8
3fffe9b0:  00000006 00000001 00000004 06190619
3fffe9c0:  00000000 00000000 00000000 00000000
3fffe9d0:  30003b4f 40266405 3fff00f8 4026a06c
3fffe9e0:  00000000 4026a081 3fff00c8 00000000
3fffe9f0:  3fff021a 3fff00d8 00000001 00000002
3fffea00:  3fff00c8 60000600 60000200 00000000
3fffea10:  40265d77 00000000 00000000 3fff021a
3fffea20:  40105d8e 60000e00 ffffffff ffffffff
3fffea30:  4026ad88 ffffffff ffffffff ffffffff
3fffea40:  00000001 b200db8e b7dc4f22 ffffffff
3fffea50:  00000000 000084c4 3fff4f3c 4026060b
3fffea60:  3ffefc2e 60000e00 3ffefc0c 3ffefc2e
3fffea70:  00000001 00000001 00000000 ffffffff
3fffea80:  40260ee3 3fff4f3c ffffffff 40260e8b
3fffea90:  3fff4f1c 00001000 401048fe 000000fd
3fffeaa0:  000000ed 00000020 3fffff10 00000001
3fffeab0:  3fff4fbc 00000001 00000003 00000000
3fffeac0:  ffffffff ffffffff ffff0002 00000000
3fffead0:  00000000 00000000 00000000 00000000
3fffeae0:  00000000 00000000 00000000 00000000
3fffeaf0:  ffffffff 00ffffff 00000000 00000000
3fffeb00:  00000000 00000000 00000000 00000000
3fffeb10:  00000000 00000000 00000000 00000000
3fffeb20:  00000000 00000000 00000000 00000000
[... 307 lines cut ...]
3ffffe60:  3fffff20 00000007 f7320f35 40001e4b
3ffffe70:  3ffffea0 00000000 00000000 00000000
3ffffe80:  be90188f d4a5e1f3 d2d8105f d607d2aa
3ffffe90:  00000000 3fffff00 3fffff00 3fffff2e
3ffffea0:  40002514 3fffdd3c 00000001 0000000d
3ffffeb0:  3fffff20 00000008 00000008 00000001
3ffffec0:  00000000 bb28d4a3 00000000 3ffffe58
3ffffed0:  00000000 519ef601 95f0a200 0e3c9f57
3ffffee0:  00000002 400042db ffffffff 4010f098
3ffffef0:  40004b31 40001878 00000001 0000000d
3fffff00:  4010f15c bb28d4a3 00000000 20000000
3fffff10:  00086b20 00000004 3fffff01 3ffe8790
3fffff20:  200304e9 40106c48 3ffe8790 00001fe0
3fffff30:  00000000 0000002d 00000000 3ffffec2
3fffff40:  4010f355 aeac1af7 3fffff50 4010f35a <
3fffff50:  00000009 3ef7c243 ffffffff 00000000
3fffff60:  d2fc7f7a db28b832 32291c9c 890e4d7c
3fffff70:  96dd2ffa 71e4b402 2a366fb8 97d1604d
3fffff80:  94989e03 c4ff6dbe d0123e2b bf8fe100
3fffff90:  97b4bb76 592a34ab a190103f fc34c031
3fffffa0:  d8c987ec 9bbc3310 f4a4bc50 d46c8846
<<<stack<<<

Exception (4):
epc1=0x400005cb epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000c depc=0x00000000

Bloodyagent, 2019-02-05:
Thanks I will try again, my issue though now is fixing the boot loop. 

So if I erase flash and flash latest bin with the esptool and it doesnt work it means the unit is bust?

Jason2866, 2019-02-05:
Yes, if powered with AC mains after a successful flash and it doesnt open an AP it has a hardware defect

Bloodyagent, 2019-02-05:
OK, so I downloaded sonoff.bin from the link you provided, and ran the following commands successfully:

esptool.py erase_flash
esptool.py --port COM4 write_flash -fs 1MB -fm dout 0x0 sonoff.bin

I'm assuming the flash command is the same for both esp8266 and esp8285?

The above has made no difference. The unit is still spewing out the above message and resetting every second and looping.

Bloodyagent, 2019-02-05:
OK, so it was a power issue!!! Board is for the first time broadcasting an SSID!! So I went into the webserver and entered my SSID and password and its sadly not connecting :), will reflash and do it again, maybe i entered incorrectly. Thanks anyway for the help to get it this far!! Will get back to you now.

Bloodyagent, 2019-02-05:
I cant believe it!! Thanks so much for your help, I have been struggling for 2 days with the Arduino IDE and here I did it in 2 mins using esptool and your help. Connected to wifi and all seems fine now. 

Thanks again

closed 2019-02-05 by Bloodyagent

Jason2866, 2019-02-05:
Glad it works