arendst/Tasmota #5126: Sonoff T1 touch switch not working

Bloodyagent, 2019-02-05:
Hi, its me again:)

I have just managed to successfully flash my T1 R2 US switch with the help of @Jason2866, thanks!

The unit is configured via web server and connected to wifi. I can toggle the switch from the web interface, however the blue touch button on the front  glass panel is not working at all. I have tried version 5.14 and 6.4.1 of the firmware will not luck. Is there a configuration setting i'm missing?

Thanks

ascillato, 2019-02-05:
Hi,

Please, could you be so kind on completing the troubleshooting template? The information required there is necessary in order to provide help. Thanks

Bloodyagent, 2019-02-05:
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Development/Compiler/Upload tools used : esptool flashing tool
- [x] Hardware used : Sonoff T1 R2 US and ftdi programmer
- [x] If a pre-compiled release or development binary was used, which one? : 5.14.0 and 6.4.1
- [x] You have tried latest release or development binaries? : yes
- [x] Provide the output of command ``status 0`` : not sure how to do this?

ascillato, 2019-02-05:
Status 0 is a command. Please, go to the Tasmota console and type status 0. Please, copy the output.

Bloodyagent, 2019-02-05:
Hi, thanks please see output below:

22:07:50 MQT: stat/sonoffT1/STATUS = {"Status":{"Module":28,"FriendlyName":["Sonoff"],"Topic":"sonoffT1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:07:50 MQT: stat/sonoffT1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:42:11","StartupUTC":"2019-02-05T20:25:39","Sleep":0,"BootCount":19,"SaveCount":55,"SaveAddress":"F5000"}}
22:07:50 MQT: stat/sonoffT1/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-07-23T16:44:37","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:07:50 MQT: stat/sonoffT1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Byterite",""],"TelePeriod":300,"SetOption":["0000A009","55818000"]}}
22:07:51 MQT: stat/sonoffT1/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":472,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
22:07:51 MQT: stat/sonoffT1/STATUS5 = {"StatusNET":{"Hostname":"sonoffT1-3767","IPAddress":"192.168.0.45","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.250","Mac":"DC:4F:22:DB:8E:B7","Webserver":2,"WifiConfig":2}}
22:07:51 MQT: stat/sonoffT1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DB8EB7","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:07:51 MQT: stat/sonoffT1/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 05 21:07:51 2019","Local":"Tue Feb 05 22:07:51 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":1,"Sunrise":"08:14","Sunset":"17:53"}}
22:07:52 MQT: stat/sonoffT1/STATUS10 = {"StatusSNS":{"Time":"2019-02-05T22:07:51"}}
22:07:52 MQT: stat/sonoffT1/STATUS11 = {"StatusSTS":{"Time":"2019-02-05T22:07:52","Uptime":"0T00:42:13","Vcc":3.230,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Byterite","RSSI":70,"APMac":"FC:EC:DA:14:C3:CB"}}}

ascillato, 2019-02-05:
Your config seems ok. It is an old version but should work fine.

Have you connected the wires of the front panel board?

Bloodyagent, 2019-02-05:
Hi, not sure what you mean by wires on the front panel? I have AC connected, including neutral. The relay activates and corresponding led turns on when i issue a mqtt command or toggle from the web interface, but the front round blue touch button does nothing.

Bloodyagent, 2019-02-05:
oh, the unit does beep when i hold the touch button down, so i think the touch sensor itself and hardware is working

Bloodyagent, 2019-02-05:
i deliberately flashed the old version to make sure it wasnt 6.4.1, but made no difference

ascillato, 2019-02-06:
Before flashing Tasmota, did you test the device? It was working?

Bloodyagent, 2019-02-06:
Good question, no I didn't! Is there a way to go back to original firmware? a download link?
What you saying, and from the fact that I cannot found any help on it, there are no configs to make this work and the unit is probably faulty?

Jason2866, 2019-02-06:
No dl link, if you didnt saved it with esptool before...

ascillato2, 2019-02-06:
Ok. So, if you have set the correct module in module config and the relays works but the touch does not work, can be a faulty hardware or your device is a clone and the gpio config for that device is different than the original sonoff touch.
You need to do some reverse engineering to see which gpios of the esp8266 are being used in your device.

Bloodyagent, 2019-02-06:
Howzit, thanks for the response guys. 

I bought it from Banggood, looks 100% original inside and out when I compare to pics online. 
I will close this issue now as there's not much more you guys can help with.
Just one more question:)
In the module parameters where you set the module type, there are these options which are all set to NONE.
GPIO1 Serial Out	
GPIO2	
GPIO3 Serial In
Do i need to do anything with these? If not then what are they for?
Thanks so much for help.

ascillato, 2019-02-06:
Those are extra pins that can be used yo hookup a sensor, switch, etc.

Bloodyagent, 2019-02-06:
Great, thanks, chat the next time I have an issue:)

closed 2019-02-06 by Bloodyagent