arendst/Tasmota #5128: mh-z19b should send temperature by mqtt too

Shmigger, 2019-02-05:
mh-z19b sensor get temperature and CO2 values by tasmota well, but tasmota do no send temperature value by mqtt, only CO2 value is sent. It would be nice to use temperature values of mh-z19b without need of additional ds18b20. and thank you for excellent project!

Jason2866, 2019-02-05:
It does 
`22:32:49 MQT: tele/sonoff-71C254/SENSOR = {"Time":"2019-02-05T22:32:49","MHZ19B":{"CarbonDioxide":836,"Temperature":21.0},"TempUnit":"C"}`

Shmigger, 2019-02-06:
Oh, I'm really sorry for my wrong topic. The problem is Mqtt for **Domoticz** would not send temperature values.

14:17:43 MQT: domoticz/in = {"idx":1,"nvalue":440,"Battery":85,"RSSI":10}
14:17:43 MQT: tele/DVES_0073FA/SENSOR = {"Time":"2019-02-06T14:17:43","MHZ19B":{"CarbonDioxide":440,"Temperature":23.0},"TempUnit":"C"}

ascillato2, 2019-02-08:
Added. Thanks :+1:

closed 2019-02-08 by ascillato2

Shmigger, 2019-02-08:
You are awesome! Thank you!!!