arendst/Tasmota #5129: Changing HOSTNAME in 6.4.1 Sonof basic Tasmota

scargill, 2019-02-05:
I'm using up to date Tasmota 6.4.1 and sometimes when I change the host name - particularly if I get it wrong in the first place - it does not seem to get completely over-written.

Example after hard-naming the host, I went back to %s-%04d in thw WIFI setup hostname - but the actual name that it is using as the hostname instead of sonoff-7051(in my case) is sonoff-7051-7051 - my network is seeing sonoff-7051-7051.broadband. Under WIFI I now have %s-%04d and in configuration-other I have "other light" -  where is it getting that second set of digits from??

ascillato2, 2019-02-06:
Hi,

Seems that some data got corrupt in your flash. Do a reset 5 (write this command in the Tasmota console) to erase all settings except wifi credentials to start over with your config. This will solve your issue. 

Remember that it is available the Tasmota Support Chat for addressing this type of issues. Github issues are for bug reporting. Thanks.

closed 2019-02-06 by ascillato2

ptranter, 2019-11-10:
I have just experienced the same problem, a repeated number on the hostname (derived from MAC address).  This does seem to be a bug, as it is consistent behaviour.
For me, it happened just after upgrading the firmware to 6.7.1.  I was changing the topic from sonoff to sonoff-xxxx, not the hostname.  A full reset configuration seemed to clear it, but it came back after I again tried to modify the MQTT topic by adding the 4 digit number to it to make it unique.
Impressive software, though - just what I needed,  Thanks