arendst/Tasmota #5130: Sensor DHT22 not working with Sonoff RF R2 Power V1.0 (new Sonoff Basic)

jacoscar, 2019-02-05:
Hello,

I'm using one of the new Sonoff Basic with the wires instead of the thick solder tracks, recognisable by the words "Sonoff RF R2 Power V1.0" printed on the board.

I've bought a DHT22 and connected the first pin to 3V3, second pin to Rx (GPIO3) and fourth pin to GND; In settings I've set "GPIO3 Serial In" to "SI7021", restarted the Sonoff but temperature and humidity are not displayed. I've also tried selecting "AM2301", and also using Tx (GPIO1) by connecting the sensor there.

The sensor works fine when connected directly to the Raspberry with both 5V or 3V inputs

Jason2866, 2019-02-05:
Have you disabled serial logging? If not you cant use rx tx

jacoscar, 2019-02-06:
Yes, It's off

```
08:31:36 CMD: SerialLog
08:31:36 RSL: stat/sonoff/RESULT = {"SerialLog":"0 (Active 0)"}
```

ascillato, 2019-02-06:
Hi,

Please, could you be so kind on completing the troubleshooting template? The information required there is necessary in order to provide help. Thanks

jacoscar, 2019-02-06:
> Hi,
> 
> Please, could you be so kind on completing the troubleshooting template? The information required there is necessary in order to provide help. Thanks

I have updated the first post

ascillato2, 2019-02-06:
It is needed a resistor for DHT22. Did you connect it?

jacoscar, 2019-02-06:
I read most people don’t use it, but I’ve tried with a 10k Ohm resistor and it didn’t do anything 
I also tried to connect it to a Sonoff S26, still using serial Rx with no luck

Jason2866, 2019-02-06:
Do you know that the sensor is working?

jacoscar, 2019-02-06:
> Do you know that the sensor is working?

It works if directly connected to the raspberry; it shows up in Home Assistant

ascillato2, 2019-02-06:
Please, try the latest full version from the development branch: http://thehackbox.org/tasmota/020500/sonoff.bin

jacoscar, 2019-02-06:
I did the OTA update, isn't it the same?

ascillato, 2019-02-06:
You have:

"Version":"6.4.1(basic)" 

I'm asking you to please try latest from development branch (http://thehackbox.org/tasmota/020500/sonoff.bin):

"Version":"6.4.1.14(sonoff)"

jacoscar, 2019-02-07:
Wow, it works now!

ascillato2, 2019-02-07:
Closing this issue as it has been solved. Thanks

closed 2019-02-07 by ascillato2

jacoscar, 2019-02-07:
I would like to understand, was I supposed to use a development version instead of the released one from the beginning?

ascillato, 2019-02-07:
You were using the **sonoff-basic.bin** that does not have sensors support. Now you are using the full version **sonoff.bin**