arendst/Tasmota #5132: CT command not recognized by Generic device

mmazojr, 2019-02-06:
Hi, 

I'm using the Generic device configuration to configure a LED RGBWW lightbulb (tuya device).  Everything seems to work fine,  except that CT commands are not recognized. I suspect it's because I'm using generic device and probably the command is not implemented for it, but I'm new to TASMOTA, so I may be completely wrong. 

If I'm right,  could someone add support for that command CT (color temperature)?  Or give me a pointer on where could I modify the code myself to add such support? 

Thanks!

ascillato2, 2019-02-06:
Hi, please, complete the troubleshooting template in order to have more information so as to properly help you. Also share an screenshot of the gpio configuration or use command gpio in console.

Jason2866, 2019-02-06:
Hi, maybe this open PR is interesting https://github.com/arendst/Sonoff-Tasmota/pull/4991

ascillato2, 2019-02-07:
Closing this issue as it as there is no feedback.

Please, if you need help, just ask here or in the Tasmota Support Chat. Thanks

closed 2019-02-07 by ascillato2

mmazojr, 2019-02-08:
Sorry for the delay. 
I'll check the suggested thread carefully and come back if I find no answer. 

Anyhow, for completeness here is a log of the problem (including a call to GPIO), the problem is that the CT command is not recognized. 

```
00:00:00 Project sonoff Sonoff Version 6.4.1(classic)-2_4_2
00:00:00 WIF: Connecting to AP1 MALMNET_2GEXT in mode 11N as sonoffrgb-2971...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoffrgb-2971.local with IP address 192.168.178.47
09:14:57 MQT: Attempting connection...
09:14:59 MQT: Connected
09:14:59 MQT: tele/sonoffrgb/LWT = Online (retained)
09:14:59 MQT: cmnd/sonoffrgb/POWER = 
09:14:59 MQT: tele/sonoffrgb/INFO1 = {"Module":"Generic","Version":"6.4.1(classic)","FallbackTopic":"cmnd/RGB_fb/","GroupTopic":"sonoffs"}
09:14:59 MQT: tele/sonoffrgb/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoffrgb-2971","IPAddress":"192.168.178.47"}
09:14:59 MQT: tele/sonoffrgb/INFO3 = {"RestartReason":"External System"}
09:14:59 MQT: stat/sonoffrgb/RESULT = {"POWER":"ON"}
09:14:59 MQT: stat/sonoffrgb/POWER = ON
09:15:07 MQT: tele/sonoffrgb/STATE = {"Time":"2019-02-08T09:15:07","Uptime":"0T00:00:15","Vcc":3.484,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":203,"POWER":"ON","Dimmer":73,"Color":"00000012","HSBColor":"0,0,0","Channel":[0,0,0,7],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"MALMNET_2GEXT","BSSId":"8C:3B:AD:17:25:CE","Channel":1,"RSSI":100}}
09:15:48 CMD: GPIO
09:15:48 MQT: stat/sonoffrgb/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"37 (PWM1)","GPIO5":"40 (PWM4)","GPIO12":"38 (PWM2)","GPIO13":"0 (None)","GPIO14":"39 (PWM3)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
09:15:51 CMD: CT
09:15:51 MQT: stat/sonoffrgb/RESULT = {"Command":"Unknown"}
```

mmazojr, 2019-02-08:
Problem solved! Thanks for the pointer @Jason2866 Checking that thread I noticed I missed defining my PWM5, adding it (I guessed the GPIO) solved the issue. 

Just for completeness of the info, these are the lightbulbs:

https://nl.aliexpress.com/store/product/2pcs-Smart-Home-Automation-E27-9W-Led-Light-Wifi-Bulb-RGB-WW-Dimmable-Remote-Control-LED/1083560_32946974132.html?spm=a2g0z.12010612.8148356.21.c3d66824jxdnzL

I flashed them OTA with tuya-convert. And setting the configuration as in the screenshot gets all the functions of the lightbulb running :)

[image]