arendst/Tasmota #5141: Reading temp/humidity from SonoffTH10 Si7021 sensor via http request nodejs

muscaiu, 2019-02-06:
I have a Sonoff Th10 with this config and i'm trying to read temp and humidity from a Si7021 sensor.

[image]

Using NodeJS and the only way to communicate with the Sonoff i found is via http requests (no libraries worked, if you use any please suggest).

These commands work fine for switching on, off and gettting the status:
http://ip/cm?cmnd=Power%20On 
http://ip/cm?cmnd=Power%20off
http://ip/ay

But, trying to get temperature and humidity data with this command i found [http://ip/cm?cmnd=status%2010](http://ip/cm?cmnd=status%2010) , all i'm getting is 

```
{
  StatusSNS: {
    Time: "2019-02-06T21:46:38"
  }
}

```

Can i get some help me with the correct config and correct http request for temperature, umidity.
Thank you!

jziolkowski, 2019-02-06:
what is the output of the same command in the console?

ascillato, 2019-02-07:
Do you see the temp value in the main Tasmota screen? If not, your sensor is not plugged in correctly

ascillato2, 2019-02-07:
Please check https://github.com/arendst/Sonoff-Tasmota/issues/4157

muscaiu, 2019-02-07:
Thanks for the quick reply and sorry for replying so late.
Below i made screens to try to aswer your questions.
If you have any more advices please let me know.
In the meantime i'm checking #4157

-----
> 
> 
> what is the output of the same command in the console?**
[image]
-----
> 
> 
> Do you see the temp value in the main Tasmota screen? If not, your sensor is not plugged in correctly

[image]
-----
[image]

muscaiu, 2019-02-07:
PS: every once in a while i get a message `DHT: Timeout waiting for start signal high pulse`
Is this related to the sensor?
```
20:37:08 MQT: Attempting connection...
20:37:14 MQT: Connect failed to 192.168.1.11:1883, rc -2. Retry in 10 sec
20:37:14 HTP: Configuration
20:37:22 HTP: Main Menu
20:37:23 DHT: Timeout waiting for start signal high pulse
20:37:24 HTP: Console
20:37:26 MQT: Attempting connection...
```

muscaiu, 2019-02-07:
HA! I did upgrade and i got the Humidity and Temp showing up on the main page.
But still it's not reading it.
Any advice?

[image]

muscaiu, 2019-02-07:
Now i'm getting a lot more messages `Timeout waiting for start signal high pulse` in the console.
```
21:10:40 MQT: Attempting connection...
21:10:45 MQT: Connect failed to 192.168.1.11:1883, rc -2. Retry in 10 sec
21:10:45 DHT: Timeout waiting for start signal high pulse
21:10:47 DHT: Timeout waiting for start signal high pulse
21:10:49 DHT: Timeout waiting for start signal high pulse
21:10:51 DHT: Timeout waiting for start signal high pulse
21:10:53 DHT: Timeout waiting for start signal high pulse
21:10:55 DHT: Timeout waiting for start signal high pulse
21:10:55 WIF: Checking connection...
21:10:55 WIF: Connected
21:10:56 MQT: Attempting connection...
```

muscaiu, 2019-02-07:
The damn jack wasn't pushed all the way in lol.

[image]

closed 2019-02-07 by muscaiu