arendst/Tasmota #5144: Do timed events occur when there is no NTP?

jpadie, 2019-02-07:
A quick question that may avoid 6 hours of travel and a plane fare...  I've no way to test this remotely.  

One of my devices is controlling a rain water pump.  At the moment it just comes on every hour for 2 minutes.  

`
rule on Time#Minute|60 do backlog power on; ruletimer1 120 endon on Rules#Timer1 do power off endon

`
This device has weak RSSI as it is under a manhole cover.  It was communicating ok when I left home but has now stopped.   My assumption is that atmospherics have changed to cause a comms drop out.  

Question:  if using the default compiled binary (which defaults to WIFI_RETRY I think) will the timed events fire every 60 minutes even if there is no time sync?  What if there has been timesync but connectivity is lost for an indeterminate time?

p.s. on a separate subject I have always found it odd is that the Sonoffs on my land (rather than in my house) don't communicate well even though they have good LoS to the access points and they are less than 20m away in clear air.  The only intervening solid object is a window with triple glazed glass.  Put the same sonoff even five metres away and its RSSI is still bad.  Put it indoors and it's fine.  Any thoughts?  Lots of different builds so there is no version consistency.

localhost61, 2019-02-07:
I make some tests and tell you...

jpadie, 2019-02-07:
Merci Xavier!  le sonoff se trouve dans un regard dans le Tarn et j'anticipe qu'il va pleuvoir plus tard!

ascillato, 2019-02-07:
>will the timed events fire every 60 minutes even if there is no time sync?

If there was no NTP sync, timers and rules will not work. 

>What if there has been timesync but connectivity is lost for an indeterminate time?

Yes, it will work until your sonoff reboots. 

You can make it more reliable adding a battery backed up RTC that it is already supported by Tasmota (DS2102)

jpadie, 2019-02-07:
> If there was no NTP sync, timers and rules will not work.

that's depressing.  Might it be better to assume that in the absence of time sync the time is 0 (unix epoch start) so that at least timed events will work on a clock cycle even if not at exactly the right time.  I guess there are edge cases that support both arguments.

localhost61, 2019-02-07:
@ascillato 
Does it come from the source code or from your experience?
Is it due to `if (RtcTime.valid)` ???

ascillato2, 2019-02-07:
If there is no time sync, Tasmota will not fire any timers because it doesn't know the time. The solution is to add an external battery backed RTC already supported by Tasmota.

closed 2019-02-07 by ascillato2

meingraham, 2019-02-07:
This coming from naivete of the code... is a hybrid mode of operation possible?

I understand that it is not possible to fire events that are timestamp based or astrologically based (sunrise/sunset) without knowing the time. Why is it not possible to still have timers that are interval based (e.g., every 60 minutes, etc.)?

Question - if the device boots and is able to sync its time, will timers continue to function even if disconnected from Wi-Fi? It retains it's time from when it first obtained it from the NTP server and goes on its merry way; yes?

Mike

ascillato, 2019-02-07:
>is a hybrid mode of operation possible?

Don't understand what you meant.

If there is wifi with NTP server, Tasmota will sync its time, so rules and timers will work.

If the device don't have wifi or a NTP server, timers and rules won't work.

If the device had synced the time and then its loose wifi or the connection to a NTP server, rules and timers will work (because Tasmota has already the time) until the device is rebooted and its loose the time. The esp8266 is that way. Don't have a battery backed RTC chip. 

If you have connected a battery backed RTC chip that is already supported by Tasmota, it doesn't matter if it has or not wifi because it will sync time with the chip. All will work.

So, 

If you have Internet connection problems, you can set a local NTP in your wifi network. 
If you have wifi connection problems, you can attach to your device a cheap battery backed RTC like DS3231.

jpadie, 2019-02-07:
MichaelI think and hope you mean astronomical rather than astrological.  It is possible to have timers firing on intervals.  Using ruletimers. E.g.  Rule on system#boot do ruletimer1 60 endon on rule#timer1 do backlog ruletimer1 86400; power on; ruletimer2 120 endon on rule#timer2 do power off endonNot tested but this should start a 60 second countdown on boot.  Then when the countdown runs out it should start the power and start two new countdowns.  One will fire after 120 seconds and turn the power off and another after 24 hours to restart the process.   This does not rely on time sync I believe; but uses millis().

localhost61, 2019-02-07:
Like for a computer, NTP is only used to sync time, so once the RTC is synced, time clock should be retained and only slightly derive of some PPM depending on the ESP8266 clock precision... till next restart.

meingraham, 2019-02-07:
Yes **astronomical** :wink: And my point here is that I can see that a network connection is required so that it can retrieve the sunrise/sunset time each day. So timers that need to know if they are a Capricorn :grin: can't work without a network connection when they need to retrieve that _astronomical_ information.

@ascillato - you answered the second part of my previous post. Once the device has a date/time, it's happy to operate any timers that are set.

Hybrid - the device would use its internal clock for interval timers (e.g., do something every N milliseconds/seconds/minutes/hours) that don't require knowing the time of day. In fact, the device DOES have a time of day even if it isn't sync'ed to a real time of day. Who cares if it thinks it's 1/1/1970 when it boots. It can still know when five minutes (or whatever interval) from that time have elapsed. So, the hybrid TASMOTA operation is that interval timers operate while timestamp timers are "disabled". In other words, a hybrid of what is now an all or nothing setup.

Mike

jpadie, 2019-02-07:
Mike

As per my post above (formatting garbled by GitHub) I think that countdown events will not require NTP.  So the rule I posted above should work.  I can't test it as I'm 1000km away from all my sonoffs I'm afraid.

localhost61, 2019-02-07:
@jpadie 
I think you're wrong due to the code I highlighted in the 4th post. And as Adrian said :

> If there was no NTP sync, timers and rules will not work.

Timers are locked if RTC is not initialized. It's maybe not the best choice but I think it's in order to keep it simple. Personally, I don't want my roller-shutters to open at sunrise if the the RTC is not synced.

jpadie, 2019-02-07:
Really?  What would be the point of blocking a countdown timer for lack of NTP?  One requires the passage of relative time whereas the other requires the knowledge of absolute time.

meingraham, 2019-02-07:
We're talking about the situation when there is potentially no MQTT transmissions thus no need to timestamp them. We're referring to a situation when one wants the switch to continue be able to control its connected device on a schedule. For example, turn a pump every hour on the hour for 5 minutes... even if the smart switch has no network connection... at that moment. Yes, the smart switch is primarily a network connected device for ultimate control and automation... but in backup mode, it would be nice to not lose all control capability. You don't need the real time of day for this.

ascillato, 2019-02-07:
@meingraham in that case, use the battery backed up RTC.

To make Tasmota use rules with and without valid RTC and to make it differentiate timed triggers from non timed ones, a major rewrite would be needed adding more code and ram usage. Sorry, that is not going to be implemented just for one marginal use-case. At least not now. We can revisit that in the future if more related issues arises. Adding the hold over label for future reference

jpadie, 2019-02-08:
@Xavier
I have briefly checked the code and ruletimers (countdowns) do not appear to be wrapped in an rtc.valid condition. 
So my suggested alternative structure would seem to work and also provide for the hybrid model that Michael was commenting about. 
My pump controller reattaches itself to the WiFi every now and then so I will try to test remotely using that (although of course I won't see the results so if anyone wants to check themselves that would be good). 

I still need to resolve the terrible WiFi connection that all my sonoffs have in open air.  It is odd - perhaps the directionality of the antenna is the issue since there are no surfaces to bounce off in the open air.  I've hacked a sonoff basic to use an external antenna so will try that when I get back home.

gishanc, 2020-12-19:
Just a quick question. To get sunrise and sunset does it need internet? Is it calculated locally or retrieved from some online API? I have a module with RTC DS3231 but very bad Wi-Fi would it work for sunset continously with no internet?

arendst, 2020-12-19:
Calculated locally based on latitude/longitude you have set.

gishanc, 2020-12-23:
Thanks a lot! Great work.

> Calculated locally based on latitude/longitude you have set.