arendst/Tasmota #5145: Devices return {"Command":"Unknown"} in reply to cmnd/sonoffs/topic, but return proper topic when issued via cmnd/<topic>/topic

jziolkowski, 2019-02-07:
Same problem across versions 6.3, 6.4.1 and 6.4 dev bins.

Other commands work, like fulltopic, restart, statuses etc. Didn't try all of them, of course, so maybe other commands don't work as well.

jziolkowski, 2019-02-07:
oh, tested it with another user, same result.

ascillato2, 2019-02-07:
Hi, I can also confirm this behaviour.

arendst, 2019-02-08:
As designed. It's a feature!

Background:
The group command (sonoffs in this case) does not allow commands like hostname to execute as it would mean that several devices would receive the same hostname which will probably make your network a bit unstable.

The same is true for some other commands although this 'group ban' has been lifted from most command over the last year.

To find out which commands are currently under control of this feature search the source code for ``grpflg``

ascillato2, 2019-02-08:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-08 by ascillato2

jziolkowski, 2019-02-08:
Ok, I get it. Could this "ban" mean they're just read-only in some future revision?

arendst, 2019-02-08:
Nice idea.