arendst/Tasmota #5148: MQTT Json Flashing

jonny190, 2019-02-07:
**Describe the bug**
When using HomeAssistant MQTT Json Lights Transition feature there is just a flash to the color then back off.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [Done] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : HomeMagic RGBW with IR remote 
- [ ] _Development/Compiler/Upload tools used_ :Master 6.4.0 / Arduino / Web Upload 
- [ ] _If a pre-compiled release or development binary was used, which one?_ : none
- [ ] _You have tried latest release or development binaries?_ : Cant Compile 6.4.1 
- [ ] _Provide the output of command_``status 0`` :
```
15:58:20 MQT: LivingroomBackdoor/stat/STATUS = {"Status":{"Module":34,"FriendlyName":["LivingroomBackdoor"],"Topic":"LivingroomBackdoor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:25","StartupUTC":"2019-02-07T14:31:55","Sleep":50,"BootCount":36,"SaveCount":240,"SaveAddress":"F9000"}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.0(sonoff)","BuildDateTime":"2019-02-07T14:28:58","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DaveysDigitalHome","Jonny190HoMeAuTo"],"TelePeriod":300,"SetOption":["00028009","558180C0","00000200"]}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS4 = {"StatusMEM":{"ProgramSize":540,"Free":460,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE7B4","000383A0","23B617CE","00003BC0"]}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS5 = {"StatusNET":{"Hostname":"LivingroomBackdoor-3059","IPAddress":"192.168.4.56","Gateway":"192.168.5.1","Subnetmask":"255.255.254.0","DNSServer":"192.168.5.10","Mac":"60:01:94:8A:CB:F3","Webserver":2,"WifiConfig":4}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.daveys.xyz","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8ACBF3","MqttUser":"sonoff","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 07 14:58:20 2019","Local":"Thu Feb 07 15:58:20 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:11","Sunset":"17:56"}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS10 = {"StatusSNS":{"Time":"2019-02-07T15:58:20"}}
15:58:20 MQT: LivingroomBackdoor/stat/STATUS11 = {"StatusSTS":{"Time":"2019-02-07T15:58:20","Uptime":"0T00:26:25","Vcc":3.439,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":0,"Color":"0,0,0,0","HSBColor":"0,0,0","Channel":[0,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"DaveysDigitalHome","BSSId":"66:70:02:CB:FE:E9","Channel":1,"RSSI":96}}}
```

**To Reproduce**
_Steps to reproduce the behavior:_
Ran script in HomeAssistant:
- service: light.turn_on
  data:
    entity_id: light.backdoor_light
    white_value: 0
    brightness: 255
    transition: 10
    rgb_color: [255,0,0]

**Expected behavior**
_A clear and concise description of what you expected to happen._
Transition from off to RED 255 in 10 secconds 

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._
Updated to allow white
GPIO_PWM4,        // GPIO13 RGBW LED White (optional - set to PWM4 for Cold White or Warm White as used on Arilux LC10)

Console output from the:
16:01:56 MQT: LivingroomBackdoor/tele/STATE = {"Time":"2019-02-07T16:01:56","Uptime":"0T00:30:01","Vcc":3.439,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":0,"Color":"0,0,0,0","HSBColor":"0,0,0","Channel":[0,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"DaveysDigitalHome","BSSId":"66:70:02:CB:FE:E9","Channel":1,"RSSI":96}}
16:01:56 MQT: LivingroomBackdoor/stat/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"0,0,0,0","HSBColor":"0,0,0","Channel":[0,0,0,0]}
16:01:56 MQT: LivingroomBackdoor/tele/STATE = {"Time":"2019-02-07T16:01:56","Uptime":"0T00:30:01","Vcc":3.444,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":20,"POWER":"ON","Dimmer":100,"Color":"255,0,0,0","HSBColor":"0,100,100","Channel":[100,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"DaveysDigitalHome","BSSId":"66:70:02:CB:FE:E9","Channel":1,"RSSI":96}}
16:01:56 MQT: LivingroomBackdoor/stat/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,0,0,0","HSBColor":"0,100,100","Channel":[100,0,0,0]}
16:01:56 MQT: LivingroomBackdoor/tele/STATE = {"Time":"2019-02-07T16:01:56","Uptime":"0T00:30:01","Vcc":3.440,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER":"OFF","Dimmer":0,"Color":"0,0,0,0","HSBColor":"0,0,0","Channel":[0,0,0,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"DaveysDigitalHome","BSSId":"66:70:02:CB:FE:E9","Channel":1,"RSSI":94}}
16:01:56 MQT: LivingroomBackdoor/stat/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"0,0,0,0","HSBColor":"0,0,0","Channel":[0,0,0,0]}
16:02:00 MQT: LivingroomBackdoor/tele/UPTIME = {"Time":"2019-02-07T16:02:00","Uptime":"0T00:30:05"}

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-02-07:
This is a question related to HA not Tasmota. Please ask for help in HA community

jonny190, 2019-02-07:
All my other mqtt transitions are working on alternative firmware to
sonoff. I'm wanting to change them all to sonoff for ease of management but
this issue is stopping me. For reference the working firmware is
https://github.com/mertenats/Arilux_AL-LC0X

ascillato, 2019-02-07:
Please, try the command weblog 4 in the Tasmota console for more debug information and then try again. Please, post the output of the console here. Thanks

ascillato2, 2019-02-07:
In the log it can be seen that the flash you see is being commanded by home assistant. So, it is not Tasmota. Is the config in Home Assistant that is sending those commands. 

See it sends:
Color2 255,0,0 (change to red without modifying the dimmer value) 
Dimmer 100 (put the bright to maximum)
White 0 (turn off all channels and put the white in off)

So Tasmota is responding ok to the commands received. Please, check the wiki for proper configuration of home assistant. 

If you need further assistant please do not hesitate on asking here or in the Tasmota Support Chat. Thanks

jonny190, 2019-02-07:
I can see the difference in the commands being send, Does tasmota not support HomeAssistant autoconfig for JSON formatting ?

Working:
Fireplace/json/set
{"state": "ON", "color": {"r": 255, "g": 0, "b": 0}, "transition": 10, "brightness": 255, "white_value": 0}
Fireplace/json/state
{"state":"ON","brightness":255,"white_value":0,"color":{"r":255,"g":0,"b":0}}

Not Working:
LivingroomBackdoor/cmnd/Color2
255,0,0
LivingroomBackdoor/cmnd/Dimmer
100
LivingroomBackdoor/cmnd/White
0

ascillato, 2019-02-07:
In the other json you see a transition command.
In the Tasmota one you show, there isn't.

ascillato, 2019-02-07:
Tasmota don't support transition from one color to another. It supports only fade.

jonny190, 2019-02-07:
My Homeassistant config for the Tasmota comes from the autoconfig "Setoption19 1"

ascillato, 2019-02-07:
Ok. So may be it is missing the fade command. Anyway, you can make an automation sending the commands with a backlog.

ascillato, 2019-02-07:
@emontnemery any idea?

emontnemery, 2019-02-08:
I'll have a look tomorrow.

ascillato, 2019-02-08:
@jonny190

Hi, could you do the test explained by @emontnemery?

jonny190, 2019-02-09:
> 
> 
> @jonny190
> 
> Hi, could you do the test explained by @emontnemery?

Ive tired this the light turnes on at 100% with not fade

ascillato2, 2019-02-12:
Can you share your modified HA config and also the output of the console?

emontnemery, 2019-02-12:
Fade is not supported by HA mqtt light with basic schema, so it's not a bug.

We could consider again adding support for HA mqtt light with JSON schema to Tasmota, which supports fading, this is a small effort.

@arendst, @ascillato2 If you will accept such PR, I can implement it. It was rejected before because Tasmota already has too many ways to control lights. But the question is popping up several times..

Another way is to make custom Tasmota component for Hass, but it would need someone to actively support it.

Jason2866, 2019-02-12:
Just my 2 cents. I dont like the idea to do specific enhancements just for HA.

emontnemery, 2019-02-12:
@Jason2866 So I can count you in on supporting the HA custom component then? ;)

jonny190, 2019-02-12:
Is the mqtt JSON transition not standard code that can be used outside of HA as well?

emontnemery, 2019-02-12:
@jonny190 It's invented by HA developers, not sure if it's used outside.

closed 2019-03-14 by stale[bot]

Jason2866, 2019-12-03:
No.