arendst/Tasmota #5149: WebThing API support

blackfyre, 2019-02-07:
Mozilla is hard at work on implementing an IOT standard which is the WebThing API.
It would be great if one of (if not) the biggest open source firmware alternative could support this!
The documentation for what the sonoffs and shellys provide as services is mature enough for implementation.

I've also made a sample implementation for a Sonoff Basic: https://github.com/blackfyre/sonoff-basic-webthing

API: https://iot.mozilla.org/wot/
GateWay: https://iot.mozilla.org/gateway/

The GateWay can be set up for testing through docker.

Jason2866, 2019-02-07:
Keep in mind the resources of the ESP devices. 
The "core" of Tasmota is mqtt. It is not the goal to do Home Automation via other protocols.
Since it is Open Source feel free to fork and do it.

ascillato2, 2019-02-07:
Hi,

Thanks for sharing your ideas. 

Adding the label hold over for future reference.

closed 2019-02-07 by ascillato2