arendst/Tasmota #5150: Sonoff Dual Amazon Alexa issue

scargill, 2019-02-07:
Firstly, Tasmota works well with Sonoff Basic Rom - but the DUAL R2 - I set that up in config - and it turns both oputputs on and off, BUT despite in config- other asking for two friendly names - and accepting them - the second one does not work with my Amazon DOT...   other Tasmota inc Sonoff BASICs do- as does the first friendly name.

I've tried a few variations - "first light","second Light", aldo "dual1","dual2"... only the first one works with Alexa.

HELP.

ascillato, 2019-02-07:
Hi, please, address this in the support Tasmota chat. Thanks

ascillato2, 2019-02-07:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-07 by ascillato2

scargill, 2019-02-08:
But - it IS a bug. Sonoff DUAL R2. Amazon Alexa - second friendly name is being ignored. But same device, same time, differrent name - node-red-contrib-amazon-echo DOES handle both names.

ascillato2, 2019-02-08:
Hi,

Please, don't get mad. 

I told you about the chat because seems to be a problem with your config and as the troubleshooting template was not completed, there is a lot of information missing in order to properly help you (like Tasmota version, status 0, console output using weblog level 4, selfcompiled or precompiled bins, etc etc) 

Besides, this was already addressed in issues.

So, please, complete the troubleshooting template, use command weblog 4 for more debug information and also share the output of the console when you are trying to switch relay1 and relay2 using alexa.
Also share the output of command gpio.

scargill, 2019-02-08:
I'm not mad - frustrated - we'll put this on hold until i resolve some network issues here, some of the tasmota sonoffs have ended up on a net I didnt even think I had... 192.168.10.x (I am 192.168.14.x) - I am of course grateful for your feedback....

scargill, 2019-02-08:
Meanwhile - answers for you - Tsamota current version up to date this week and I'm using your release bins...

meingraham, 2019-02-08:
Pete, Thanks for the version information. However, "current version" and "your release bins" still doesn't narrow it down much. There are still variables like which ESP core and which flavor of the bins (basic, sensors, etc.). That is why providing the output of STATUS 0 is requested. It provides a lot of details in a concise output. Not only does it give the specific version information, but it also provides the information on what settings you have selected. All of that information help the community help you... and hopefully get you to the point where this is resolved and reduce your frustration.

ascillato2, 2019-02-08:
>Meanwhile - answers for you - Tsamota current version up to date this week and I'm using your release bins...

Sorry, what we need when opening an issue is the template completed. Please, use it. It is not a silly bourocratic thing. It is very useful guide for requesting the most common information.
At this point I don't know if you are using Tasmota v6.4.1 or 6.4.1.14, and a lot of other things.

Besides, to solve a bug we need much more information to reproduce it and to have a hint what and where to fix.

At this moment I couldn’t reproduce your issue. A sonoff dual R2 is working fine with Alexa. That is why we need more information to troubleshoot your issue. We want yo help, but without information we will be just guessing.

Sorry, in order to receive help, you need to provide information.

scargill, 2019-02-08:
I've gone to the discussion area as requested as I'm looking for help (it is usually me providing help for others).

meingraham, 2019-02-09:
That's probably not a bad idea. Based on my experience here, on the Discord channel, and on the Sonoff Users Google group, the question about specific versions and configurations (i.e., STATUS 0 and GPIO), are usually part of the dialog. I'm sure that when you're the one on the giving end of the assistance, you usually ask questions to help understand the situation so you can provide informed advice. No different here. Here's hoping you get the help you seek.

Cheers!

Mike