arendst/Tasmota #5151: Color/PWM inconsistency issue collection

armageddon421, 2019-02-07:
**Describe the bug**

There is no way to accurately set RGB/RGBW values, nor is there a way to set multiple PWM values at once to achieve a desired result. I've collected some unresolved but closed issues describing the various problems.

- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : irrelevant, all hardware is affected
- [ ] _Development/Compiler/Upload tools used_ :
- [x] _If a pre-compiled release or development binary was used, which one?_ : latest sonoff_DE from releases
- [x] _You have tried latest release or development binaries?_ : yes
- [ ] _Provide the output of command_``status 0`` : please see below issues for details

**To Reproduce**
repeatedly set channel 2, other channels decrease each time
issue: #4558 
```
'stat.dhaled3.color':'F8007D'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F6F97C'
'cmnd.dhaled3.channel2':'98'
'stat.dhaled3.color':'F5F97B'
...
```

issue: #4901 
Delay between backlog commands of 50ms is definitely visible on an RGB or RGBW stripe, so there's no way to set multiple PWM channels at the same time in setOption15 0 mode.
Also, using that mode would completely lose any ability to dim or fade, so it's not really an option.

issue: #4797 
describes the accuracy error due to "rounding from 255 to 100 to 255", to reproduce

> transmit via MQTT the color (201,204,0) the console shows
> `{"POWER":"ON","Dimmer":80,"Color":"200,204,0","HSBColor":"61,100,80","Channel":[78,80,0]}`

duplicate issue: #4803 
> Behaviour seen on H801.
> Start with all channels 0.
> Set 1st channel by "channel1 30" -> response includes "Channel":[28,0,0,0,0]
> Set 2nd channel by "channel2 30" -> response includes "Channel":[27,28,0,0,0]
Here you can see both problems. The value of 30 is not accuarately set, and channel1 is influenced by setting channel2

**Expected behavior**
There should be a consistent way to reliably and accurately set the RGB(W) values with full resolution of 0 to 255, without influencing other channels in case only one channel was changed.

**Additional context**
- One use case is where I want to use the separate channels for separate white lights, so only one might be changed at a time, while still using fading (hence not using legacy PWM mode). The channels should be independent and not be influenced by the other channels, especially not continually decreasing.
- I'd like to be able to set a color value of as low as "1" to allow for very dim lights at night, while using the full potential of a relatively bright light at day/in the evening. Effectively having only 100 brightness levels greatly reduces color resolution in dark settings.
- When setting a color via any UI and also updating the UI when an update is received from the tasmota device, the user should not be presented with a value different to what he just set. This also prevents any logic along the lines of "if light is not on default value, reset it after x minutes", for example when automating a light but still allowing manual control, because it will never take that exact default value.

I hope I could provide some insight into the general area of these issues and how they actually cause trouble, can be considered bugs and not features and what can be gained by improving upon this topic. Personally, I am experiencing all of these issues and so far could only effectively work around some of them.
Otherwise, Tasmota has been an amazing experience for me so far and I really appreciate all the hard work that went into this. Thank you.

Jason2866, 2019-02-07:
Keep in mind the resources of a ESP device. You cant do all these things without making compromises

armageddon421, 2019-02-07:
Well, my guess is that the complexity high right now and some of the code just needs a little cleanup, maybe some small paradigm changes like increasing dimmer resolution to 255 or simply not touching the RGB values if the dimmer changed due to directly setting the RGB values. Except from maybe more complicated fading, this should even reduce the required resources.

Jason2866, 2019-02-07:
See comment from project owner for changing light driver https://github.com/arendst/Sonoff-Tasmota/pull/4991#issuecomment-457513830

armageddon421, 2019-02-07:
That issue is about devices with separate chip to drive RGB but then another pwm channel directly attached to the esp8266 to controls the white channel. I understand the complications of that, but it has few to do with the general issues outlined here as they are situated above the driver level.

andrethomas, 2019-02-07:
@arendst may respond to this one, but currently, the light driver is a bit of a challenge insofar that new types of devices are being discovered by Sonoff-Tasmota users so until there is a simpler way to handle the complexities of introducing support for new devices this will most likely not be a priority.

arendst, 2019-02-08:
It's currently as designed.

New hardware pops up like the measles in US and there were some decisions made in the past that influence the future.

Regarding changing a dimmer value while keeping RGB untouched I suggest you dive into the intricacies of RGB lights to find that dimming is nothing more than lowering the RGB values.

Feel free the provide solutions to your issues while keeping (and proven tested) backward compatibility as they would be accepted almost without discussion.

armageddon421, 2019-02-08:
> Regarding changing a dimmer value while keeping RGB untouched I suggest you dive into the intricacies of RGB lights to find that dimming is nothing more than lowering the RGB values.

I know that. As far as I understand, currently if you change the RGB values directly, a dimmer value is derived from that and the dimmer value is updated. Then the RGB values are in turn updated to match this dimmer value. The lower resolution of the dimmer value then causes the RGB values to change. The suggestion was to simply prevent this feedback unless the dimmer value was directly changed.

> Feel free the provide solutions to your issues while keeping (and proven tested) backward compatibility as they would be accepted almost without discussion.

I'll try my luck in the next days, I'm just not familiar with the codebase, so it might take some time. I'll make sure to stick to the given paradigms and keep backwards compatibility. For example if I change the dimmer to 255 resolution internally, the external interfaces would still up/downsample to a percentage.

ascillato2, 2019-02-08:
@armageddon421 Thanks for grouping all those issues. That is very useful for issue tracking. The label requested feature has been added for future reference.
Also this issue has been referenced to the actual open PR requesting a change in the light driver. 

We will be waiting for your ideas for the code :+1:

closed 2019-02-08 by ascillato2

armageddon421, 2019-02-11:
I've investigated the code and I think I found most of the causes of these issues.
One suggestion would be to store any user input values like a color command as-is instead of normalizing it to fill the available range and use the dimmer to scale it back. Instead, premultiply the values each time before applying the dimmer. I already tried removing the normalization step from the LightSetColor function and instead setting the dimmer to 100 in that case, and most of the problems seem to be gone.
This will lead to the following improvements/compromises:
- right after directly setting a hex color value, it is applied with its full resolution
- changing the dimmer (in percent) will obviously cause in a loss of resolution
- changing only one channel using the channel command will not affect the other channels
- it is not possible to return to the exact initial color/channel setting by just setting the dimmer back to the value calculated for that, because it is simply less precise.
I'll try changing the rest of the code so functionality matches the way it is right now tomorrow.
Only remaining mystery is currently why channel commands always result in one less than specified. Probably a float rounding error. I'll try to remove floats where possible, they're slow on the esp anyways.