arendst/Tasmota #5152: New Sonoff Basic with longer wires

jacoscar, 2019-02-07:
Hi all,

I'm aware there is another discussion about this topic, but the ones I've just received from Amazon are again different; they have longer wires that go over the 4 holes we need to flash them and they're pretty stiff to move, making our job even harder.

Has anybody have to deal with them before?

Jason2866, 2019-02-07:
It is the same. Just longer wires used...
https://raw.githubusercontent.com/reloxx13/reloxx13.github.io/master/media/tasmota/sonoff-basic-r2/front.jpg

jacoscar, 2019-02-07:
If I move the wires to make room for the pins then it doesn’t close
Where can I buy the first version (the one with no wires at all)?

Jason2866, 2019-02-07:
This is the latest revision from ITEAD....  You wont get the old version anymore
unsolder the long wires shorten it and solder it

meingraham, 2019-02-07:
Move the AC wires, flash the device, move the AC wires back, close?

jacoscar, 2019-02-07:
> Move the AC wires, flash the device, move the AC wires back, close?

it doesn't work if you want to connect a sensor or a switch; shortening the wires seems the only solution; maybe they did it to discourage flashing?

Jason2866, 2019-02-07:
I dont think so. Probably just a worker who has problems to cut wires in correct needed length...

meingraham, 2019-02-07:
> it doesn't work if you want to connect a sensor or a switch

Can you solder the wire for the sensor directly to the PCB rather than using a pin header? Then route the wire for the sensor under the AC wires and out to the case.

> maybe they did it to discourage flashing?

It seems that they have done nothing but make changes to discourage the DIY/maker community. Shelly and Tuya-based devices would be my first choice by far these days.

@Jason2866 - that worker needs to be moved to a different kind of job if they have this much trouble cutting a wire to a certain length :wink:

Mike

closed 2019-02-07 by ascillato2