arendst/Tasmota #5157: Tasmota on NodeMCU with DS18S20 not working

vermmatt, 2019-02-07:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Arduino IDE
- [x] Hardware used : NodeMCU, DS18B20
- [x] If a pre-compiled release or development binary was used, which one? :
- [x] You have tried latest release or development binaries? :
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Hi,

I have flashed Tasmota on a NodeMCU device and everything seems to be working correctly.
The module is configured as "Generic".

Now I wanted to get a temperature reading using a DS18B20 sensor.
I connected the data end of the sensor to GPIO14 of the module and pulled VCC to data with a 4.7 K resistor.

In the Tasmota configuration I set GPIO14 to DS18x20 but nothing happens.
When the device restarts, I don't see any status lines on the DS18x20 sensor.
(When DHT11 is selected, I can see a "null" reading)
So I'm having a bit of a hard time getting this to work.

Any advice is appreciated.

Thank you

armageddon421, 2019-02-07:
I've also had trouble with some specific pins. Last time I used GPIO4 and it worked fine. Even GPIO0 worked for me in the past, if there is no button connected to it. maybe try some different pins?

ascillato2, 2019-02-07:
Hi, please for support use the Tasmota Support Chat. Github issues are for bug reporting. Thanks 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-07 by ascillato2