arendst/Tasmota #5161: Power value randomly drops to 0.0

davekr8, 2019-02-07:
**As can be seen in the attached log, there is a random drop of power values. Strange is, that both the current and voltage values are not equal to zero. There must be a calculation or sampling error in the processing.**

- [X] _Searched the problem in issues and in the wiki_
- [X] _Hardware used_ : 

**BlitzWolf SHP Module**

- [ ] _Development/Compiler/Upload tools used_ :
- [X] _If a pre-compiled release or development binary was used, which one?_ :

**Program Version	6.4.0(sonoff)
Build Date & Time	2018-12-16T14:31:02
Core/SDK Version	2_4_2/2.2.1(cfd48f3)**

- [ ] _You have tried latest release or development binaries?_ :
- [X] _Provide the output :
```
22:19:50 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:19:50","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.158,"Yesterday":1.012,"Today":1.115,"Power":0.0,"ApparentPower":210.4,"ReactivePower":210.4,"Factor":0.00,"Voltage":236,"Current":0.890}}
22:19:51 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"OFF","PowerHigh":"ON"}
22:19:51 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:19:51","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.158,"Yesterday":1.012,"Today":1.115,"Power":192.2,"ApparentPower":210.4,"ReactivePower":85.6,"Factor":0.91,"Voltage":236,"Current":0.890}}
22:24:53 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"ON","PowerHigh":"OFF"}
22:24:53 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:24:53","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.175,"Yesterday":1.012,"Today":1.131,"Power":0.0,"ApparentPower":213.7,"ReactivePower":213.7,"Factor":0.00,"Voltage":235,"Current":0.908}}
22:24:54 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"OFF","PowerHigh":"ON"}
22:24:54 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:24:54","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.175,"Yesterday":1.012,"Today":1.131,"Power":193.3,"ApparentPower":211.3,"ReactivePower":85.3,"Factor":0.91,"Voltage":235,"Current":0.898}}
22:30:47 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"ON","PowerHigh":"OFF"}
22:30:47 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:30:47","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.194,"Yesterday":1.012,"Today":1.151,"Power":0.0,"ApparentPower":212.7,"ReactivePower":212.7,"Factor":0.00,"Voltage":235,"Current":0.904}}
22:30:48 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"OFF","PowerHigh":"ON"}
22:30:48 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:30:48","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.194,"Yesterday":1.012,"Today":1.151,"Power":191.9,"ApparentPower":211.5,"ReactivePower":89.0,"Factor":0.91,"Voltage":235,"Current":0.899}}
22:40:48 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"ON","PowerHigh":"OFF"}
22:40:48 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:40:48","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.227,"Yesterday":1.012,"Today":1.184,"Power":0.0,"ApparentPower":214.6,"ReactivePower":214.6,"Factor":0.00,"Voltage":235,"Current":0.912}}
22:40:49 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"OFF","PowerHigh":"ON"}
22:40:49 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:40:49","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.227,"Yesterday":1.012,"Today":1.184,"Power":193.0,"ApparentPower":209.4,"ReactivePower":81.3,"Factor":0.92,"Voltage":235,"Current":0.890}}
22:44:16 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"ON","PowerHigh":"OFF"}
22:44:16 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:44:16","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.239,"Yesterday":1.012,"Today":1.195,"Power":0.0,"ApparentPower":208.9,"ReactivePower":208.9,"Factor":0.00,"Voltage":235,"Current":0.888}}
22:44:17 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"OFF","PowerHigh":"ON"}
22:44:17 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:44:17","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.239,"Yesterday":1.012,"Today":1.195,"Power":196.2,"ApparentPower":211.5,"ReactivePower":79.1,"Factor":0.93,"Voltage":235,"Current":0.899}}
22:49:34 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"ON","PowerHigh":"OFF"}
22:49:34 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:49:34","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.256,"Yesterday":1.012,"Today":1.213,"Power":0.0,"ApparentPower":214.8,"ReactivePower":214.8,"Factor":0.00,"Voltage":236,"Current":0.910}}
22:49:35 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerLow":"OFF","PowerHigh":"ON"}
22:49:35 MQT: home/gosund/gosund.1/tv/SENSOR = {"Time":"2019-02-07T22:49:35","ENERGY":{"TotalStartTime":"2019-01-11T21:02:36","Total":23.256,"Yesterday":1.012,"Today":1.213,"Power":198.1,"ApparentPower":214.5,"ReactivePower":82.2,"Factor":0.92,"Voltage":236,"Current":0.910}}
23:02:00 MQT: home/gosund/gosund.1/tv/UPTIME = {"Time":"2019-02-07T23:02:00","Uptime":"25T06:22:45"}
23:07:50 MQT: home/gosund/gosund.1/tv/MARGINS = {"PowerHigh":"OFF"}
```

ascillato2, 2019-02-07:
The measurement is being done in another chip different than the esp8266 and it measure each value in a different time. So, if you are measuring something with a switching power supply, this type of measurements are expected.

Sorry, there is nothing that can be done from the software side with the cheap measurement chip that device has. You can change the device to a pow R2 or use a PZEM for power measurement. 
Or measure another thing that does not have that switching power supply.

If you need further assistant please find us in the Tasmota Support Chat. Thanks

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-07 by ascillato2

davekr8, 2019-02-08:
Ok. To get you right, the powere 0.0 results from the switching power supply of my running tv, that is measured at the moment of switching. But, if this is the case the current must be 0 as well. Anyway, there must be wrong values in the sensor data representation, and current and voltage are still present.

Riwaha, 2019-03-13:
I am noticing a very similar experience, sometimes the power is a value of 0, when this happens sometimes the current is also 0 but not always. The load is a computer drawing around 150-200 watts.

tobox, 2019-06-05:
I have exactly the same problem with with a Sonoff Pow (R1) with an air conditioning (interesting line is 14:16:53):

```
14:16:49 MQT: tele/sonoff-power-klima/STATE = {"Time":"2019-06-05T14:16:49","Uptime":"58T20:55:26","Vcc":3.153,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"ToBoX","BSSId":"F0:B0:14:4F:5E:06","Channel":11,"RSSI":78,"LinkCount":5,"Downtime":"0T00:00:22"}}
14:16:49 MQT: tele/sonoff-power-klima/SENSOR = {"Time":"2019-06-05T14:16:49","ENERGY":{"TotalStartTime":"2019-04-09T00:00:00","Total":518.783,"Yesterday":6.233,"Today":6.535,"Period":3.71,"Power":884.00,"ApparentPower":890.85,"ReactivePower":110.30,"Factor":0.99,"Voltage":236.3,"Current":3.770}}
14:16:53 MQT: tele/sonoff-power-klima/SENSOR = {"Time":"2019-06-05T14:16:53","ENERGY":{"TotalStartTime":"2019-04-09T00:00:00","Total":518.784,"Yesterday":6.233,"Today":6.536,"Power":0.00,"ApparentPower":890.94,"ReactivePower":890.90,"Factor":0.00,"Voltage":236.7,"Current":3.764}}
14:16:54 MQT: tele/sonoff-power-klima/SENSOR = {"Time":"2019-06-05T14:16:54","ENERGY":{"TotalStartTime":"2019-04-09T00:00:00","Total":518.784,"Yesterday":6.233,"Today":6.536,"Power":879.50,"ApparentPower":889.28,"ReactivePower":131.50,"Factor":0.99,"Voltage":236.7,"Current":3.757}}
14:17:04 MQT: tele/sonoff-power-klima/STATE = {"Time":"2019-06-05T14:17:04","Uptime":"58T20:55:41","Vcc":3.152,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"ToBoX","BSSId":"F0:B0:14:4F:5E:06","Channel":11,"RSSI":78,"LinkCount":5,"Downtime":"0T00:00:22"}}
14:17:04 MQT: tele/sonoff-power-klima/SENSOR = {"Time":"2019-06-05T14:17:04","ENERGY":{"TotalStartTime":"2019-04-09T00:00:00","Total":518.786,"Yesterday":6.233,"Today":6.539,"Period":3.70,"Power":885.30,"ApparentPower":894.25,"ReactivePower":126.20,"Factor":0.99,"Voltage":236.2,"Current":3.786}}
14:17:19 MQT: tele/sonoff-power-klima/STATE = {"Time":"2019-06-05T14:17:19","Uptime":"58T20:55:56","Vcc":3.154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"POWER":"ON","Wifi":{"AP":1,"SSId":"ToBoX","BSSId":"F0:B0:14:4F:5E:06","Channel":11,"RSSI":72,"LinkCount":5,"Downtime":"0T00:00:22"}}
14:17:19 MQT: tele/sonoff-power-klima/SENSOR = {"Time":"2019-06-05T14:17:19","ENERGY":{"TotalStartTime":"2019-04-09T00:00:00","Total":518.790,"Yesterday":6.233,"Today":6.543,"Period":3.75,"Power":898.40,"ApparentPower":903.01,"ReactivePower":91.10,"Factor":0.99,"Voltage":236.7,"Current":3.815}}
```
It may be an error of the measuring chip, but in that case the error would also occur with the original Sonoff firmware. Unfortunately, I don't have any POWs with the original firmware left to verify.

arendst, 2019-06-05:
As power, current and voltage are measured at different timestamps it is possible one of them is 0 while the others were not zero.

I'll add a retry to the power measuring to keep the last value if the current value is 0. Let's see if that solves your issue.

tobox, 2019-06-26:
My Airconditioner has been running for many hours in the last days and I have not seen any drops to 0 since the patch. Seems to be working.