arendst/Tasmota #5163: Issue with tasmota after power outage

CancunManny, 2019-02-08:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

**Describe the bug**
_A clear and concise description of what the bug is._
I have 4 esp866 (Wemo D1 Mini) set up.  They are all flashed with tasmota 6.4.1
When power goes out (here in Cancun Mexico a bit more often than one would like) usually at least one of the tasmotas doesn't respond when power comes back up. It is not the same tasmota that gives the issue, it can be any of the 4.  Usually at least one of them connects correctly.

I had noticed the issue before, so I went back to all the tasmotas and made sure none of the special pins were being used (GPIO15, GPIO0, etc)

Today had a short power outage of about a minute.  3 out of the 4 didn't respond correctly.  For one of them it was as easy as turning the power off once then back on, and it responded.  The other two I have no "easy" way to turn power off so I had to pull the esp8266 out of its socket and back on.  Here is the weird part, on one of them I had to unplug and plug the esp about 3 times before it booted correctly (LED blinks after power is turned on instead of going solid), and on the other one I had to unplug it and plug it back in about 20 times before it responded.  I even unplugged it for about 3 minutes then plugged it back in and still wouldn't work.  While thinking on what else I could try I kept taking it out of the socket and back in until all of the sudden the LED started blinking when.

If I understand correctly, the way it should work after a power outage is as follows.  When power comes back on, the tasmotas seem to boot quicker than the router itself.  Since it can't connect to wifi after a couple of tries it goes into Access Point mode for about 2 to 3 minutes, it then checks to connect to wifi.  I have noticed that when they boot correctly the LED starts blinking, which is when it is connecting to wifi and then mqtt.  If it doesn't boot correctly the LED goes straight to solid, and the push buttons to turn on and off the lights don't respond.

Trying to provide as much helpful info so I do want to mention that I am using some cheap USB power adaptors, the type used to charge cell phones.  I know they are cheap because I tried to use touch sensors instead of push buttons and the touch sensors wouldn't work when using those cheap USB adaptors.  Would hate to have to go through the hassle of changing the USB adaptors only to find out that is not the issue.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [ x] _Hardware used_ : Esp8266, D1 Mini Wemos
- [x ] _Development/Compiler/Upload tools used_ : Arduino IDE
- [x ] _If a pre-compiled release or development binary was used, which one?_ :  Not sure, I think it was a pre compiled releas
- [x ] _You have tried latest release or development binaries?_ : I think 6.4.1 is the latest version
- [ ] _Provide the output of command_``status 0`` :

```
STATUS 0 OUTPUT HERE - DO NOT DELETE THE MARKERS ABOVE AND BELOW THIS LINE
01:50:27 MQT: CuartoMilly/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Viento","Sol"],"Topic":"CuartoMilly","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,5,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:50:27 MQT: CuartoMilly/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:20:42","StartupUTC":"2019-02-08T00:29:45","Sleep":50,"BootCount":127,"SaveCount":2107,"SaveAddress":"FA000"}}
01:50:27 MQT: CuartoMilly/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-01-02T17:01:13","Boot":31,"Core":"2_5_0_BETA2","SDK":"3.0.0-dev(c0f7b44)"}}
01:50:27 MQT: CuartoMilly/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Miriam2","Miriam"],"TelePeriod":300,"SetOption":["000A8009","55C180C0","00000200"]}}
01:50:27 MQT: CuartoMilly/stat/STATUS4 = {"StatusMEM":{"ProgramSize":708,"Free":292,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
01:50:27 MQT: CuartoMilly/stat/STATUS5 = {"StatusNET":{"Hostname":"CuartoMilly-7410","IPAddress":"10.0.0.240","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.223.234.2","Mac":"CC:50:E3:55:3C:F2","Webserver":2,"WifiConfig":2}}
01:50:27 MQT: CuartoMilly/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.227","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_553CF2F2","MqttUser":"mqttuser","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:50:27 MQT: CuartoMilly/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 08 00:50:27 2019","Local":"Fri Feb 08 01:50:27 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:10","Sunset":"17:58"}}
01:50:27 MQT: CuartoMilly/stat/STATUS10 = {"StatusSNS":{"Time":"2019-02-08T01:50:27","Switch1":"OFF","Switch2":"ON"}}
01:50:27 MQT: CuartoMilly/stat/STATUS11 = {"StatusSTS":{"Time":"2019-02-08T01:50:27","Uptime":"0T00:20:42","Vcc":3.018,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Miriam2","BSSId":"04:95:E6:DC:2A:D8","Channel":11,"RSSI":72}}}

```

**To Reproduce**
_Steps to reproduce the behavior:_
I can't reproduce the behavior at will.  At times I am able to turn power off to the whole house, then back on and they all boot correctly.  Other times some come back and others don't

**Expected behavior**
_A clear and concise description of what you expected to happen._

When power goes out and comes back, tasmotas should look for wifi connection, if it can't connect it should go into AP mode for a few minutes, then try to connect again.  Also the physical push buttons should work to turn lights on and off even if no internet connection

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-02-08:
Hi

Please, share the gpio configuration. Using an screenshot or using command gpio on the console. Thanks

CancunManny, 2019-02-08:
Hello ascillato. I do have 4 different tasmotas set up.  Like I mentioned after a power outage, at times all of them come back with no issues, while other times some don't come back.  I used the same file to flash all 4 of them so they all have the same firmware.  The GPIO configuration is a bit different in all of them, but none of them are using any of the special pins.  This is the GPIO configuration of the tasmota I used to get the status output

`03:43:16 RSL: CuartoMilly/stat/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"10 (Switch2)","GPIO3":"0 (None)","GPIO4":"22 (Relay2)","GPIO5":"0 (None)","GPIO12":"9 (Switch1)","GPIO13":"52 (Led1)","GPIO14":"21 (Relay1)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
`

ascillato2, 2019-02-08:
GPIO2 can’t be low at boot, so you can’t connect a switch to it.

See gpio restrictions at https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs

ascillato2, 2019-02-08:
You should change the gpio selection for your devices.

If you need further assistant please do not hesitate on asking here or in the Tasmota Support Chat. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-02-08 by ascillato2

CancunManny, 2019-02-08:
Hello again.  Ok so I chose the wrong example.  I just remembered that specific one was the last one I had to take down and change around.  For what it is worth, the info I shared was for "viento", and that is the room that has a separate power breaker.  That one I only had to turn the breaker off then on, and it booted without an issue after that.  here is the list of the 3 tasmotas GPIO readout

Cocina
`04:03:05 RSL: Cocina/stat/RESULT = {"GPIO0":"0 (None)","GPIO1":"11 (Switch3)","GPIO2":"0 (None)","GPIO3":"52 (Led1)","GPIO4":"21 (Relay1)","GPIO5":"9 (Switch1)","GPIO12":"22 (Relay2)","GPIO13":"10 (Switch2)","GPIO14":"23 (Relay3)","GPIO15":"0 (None)","GPIO16":"0 (None)"}`
 
ventilador
`04:08:21 RSL: SalaMiriam/stat/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"0 (None)","GPIO12":"21 (Relay1)","GPIO13":"22 (Relay2)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}`

Luz Vale
`04:08:57 RSL: CuartoVale/stat/RESULT = {"GPIO0":"0 (None)","GPIO1":"10 (Switch2)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"9 (Switch1)","GPIO5":"0 (None)","GPIO12":"21 (Relay1)","GPIO13":"52 (Led1)","GPIO14":"22 (Relay2)","GPIO15":"0 (None)","GPIO16":"0 (None)"}`

ascillato, 2019-02-08:
Ok, your module config seems ok, but in your circuit, there is something connected to gpio0, gpio2 or gpio15? 

Because your issue seems to be just a boot mode issue as explained at https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#boot-modes

If there isn't anything connected and those pins are floating, may be the issue is just noise in the power supply. The esp8266 uses a peak of power when it boots, so, if the power source is not good or can't deliver the required power, your device won't boot or will not be able to use the wifi.

CancunManny, 2019-02-08:
First, I do want to thank you for your assistance, specially since you are replying so quickly.  I spent about 3 weeks fighting with the tasmotas, learning the hard way.  I did build my own "sonoffs".  At one point I had all the components up on the celing, but since I kept having issues and I kept learning new stuff I realized that bringing the actual ESP8266 chip to the wall plate switch made more sense since they are now much more accessible.  Part of the issue I was having is that I was using some of the special pins.  That is why 3 out of the 4 tasmotas are configured without any of the special pins.  The "viento" tasmota is the only one I have to go back and switch around so that it ends up on a wall plate rather than in the ceiling.  

I am 100% sure there is nothing connected to pins GPIO 0, 2 and 15 on the 3 tasmotas.  The cheap USB power supplies are actually reading 5.3 V (without a load).  My guess is that the problem is the noise rather than not being able to supply enough power.

I did post my issue first on HomeAssistant forum, and they were quick to let me know I should post in this site instead because it is more of a tasmota issue than a HA issue.  However handful of members mentioned they had nothing but problems with version 2.4 and 2.5.  I checked and I am  using version 2.5 Beta.  The suggest i re flash using version 2.3 which they claim works flawlessly.  I am not a big fan of going "backwards" with softwear/firmware.

The 4 tasmotas I have set up at the moment are at my gf's house.  I set her house up first because I knew I was going to be moving soon.  This weekend I just moved to my new place.  I have some "real" power supplies now (HiLink). I plan to build my poor mans sonoffs in the next couple of days, and see if I experience the same issues.  If I don't have any issues with my setup I would assume the issue is the cheap USB power supplies, and I would change them out at my GFs house, and I would come back here to confirm that was the issue. 

On the other hand, if I do end up with the same issues with my setup I would have to assume is not the power supply that is causing the issue, and I would post here again.

ascillato, 2019-02-08:
Yes, at first is hard to understand all this new stuff but then it goes easy and fun. 

You are right, there are some setups (power supply, device positions, etc etc) that needs to be tested in order to find the more appropiate for each case. 

The same happens with the core. Tasmota uses the Arduino core and different versions of the core works better for some people and some others don't. That depends on the brand of your router and the type of wifi setup and config. A brief explaination of this with the recommended solutions is at:

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-arduino-core-versions-and-expressif-sdk

Hope this help you on chosing and testing the best core for you. Anyway the recommended is the latest core 2.5.0 with the latest Tasmota version 6.4.1.14

Lastly, I think that with a good power supply like HiLink your issue should be solved. 

Anyway, do not hesitate on asking here or in the Tasmota Support Chat.

budlyte56, 2019-07-12:
I have 5 sonoff basic flashed with the latest tasmota.  If there is a power failure only some of the basic's come back up and talk to the internet.  When this happens I have found the only solution is to use the button press sequence ( 4 short presses) to start it talking to the internet on 192.168.4.1.  I log in to the wifi id which will then open the wifi configuration.  All I have to do is put back in my two wifi passwords, save which restarts the device and all is well.  Am I doing something wrong.  Have all the retain setting as described by the hookup in his video.

Thanks

dharvind, 2022-02-02:
> First, I do want to thank you for your assistance, specially since you are replying so quickly. I spent about 3 weeks fighting with the tasmotas, learning the hard way. I did build my own "sonoffs". [...]

Hi @CancunManny,

Have you been able to figure it out finally?

I'm facing a similar situation after 3 months running Sonoff Tasmota 9.2.0 devices without any issue. All my 20 Sonoff devices are wired to the mains directly and today after multiple power cuts (around 6 power cuts lasting an average of 10 minutes), 5 devices did not come up. My router is connected to a UPS and hasn't powered off, so it cannot be the AP/Wifi race scenario you mentioned.

I checked the sonoff devices which are not showing up, and they are not responding, as in:
1. no effect of tripping the breaker to cause it to reboot again
2. no led lights on the device light up
3. pressing the button on the sonoff does not do anything (on the other hand on the working sonoff devices, pressing the button turns on/off the lights connected to it
4. no AP displayed which means it did not reset

@ascillato regarding the power noise you mentioned - is it possible that only few devices are affected while others are drawing power from the same source?
Electrical setup approximately like this: Mains -> Breaker 1 to 5 -> Sonoff 1 to 5 (Breaker 1 -> Sonoff 1 -> Light 1)
I did have oh so frustrating power outages today so I'm thinking this triggered the problem...

ascillato, 2022-02-02:
> regarding the power noise you mentioned - is it possible that only few devices are affected while others are drawing power from the same source?

Yes, it is possible because of the quality of the filtering at the power regulator inside the device. For example, a sonoff basic has a very poor regulator and you may expect some issues, but a shelly 1 has a better power regulator and it shouldn't have issues of this type.

ascillato, 2022-02-07:
Without soldering there isn't much that can be done to fix the power regulator and that device should be replaced. Sorry.

If you want, you can try tasmota-lite.bin.gz version that uses less energy because it hasn't extra features nor sensors support.

lalo-uy, 2022-02-08:
> Thanks @ascillato <https://github.com/ascillato>
> I think the issue is with the power regulator inside the sonoff as you
> said since I've confirmed that:
>    1. mains power is coming to the sonoff device
>    2. the sonoff device boots fine when disconnected and powered through
>    usb 3.3v
>    3. I was able to reflash the sonoff device offline and it is able to
>    connect to wifi (still through usb 3.3v)
>    4. when connected back to the mains, it is still off and does not
>    As per your experience, is this still fixable (without any soldering),
>    or is it better that I replace the device all together?

The sonoff power suplly is very easy to break.
You can check if it is working measuring the voltage at the imput of the
regulator. Shuld be about 5v.
I have several units with bourned power supply that work ok id powered with
3v

dharvind, 2022-02-11:
> Without soldering there isn't much that can be done to fix the power regulator and that device should be replaced. Sorry.
> 
> If you want, you can try tasmota-lite.bin.gz version that uses less energy because it hasn't extra features nor sensors support.

Thanks @ascillato tried the minimal version but no luck. Will replace the device.

dharvind, 2022-02-11:
> The sonoff power suplly is very easy to break. You can check if it is working measuring the voltage at the imput of the regulator. Shuld be about 5v. I have several units with bourned power supply [...]

Thanks @lalo-uy did you replace yours or repair by soldering?

lalo-uy, 2022-02-13:
It’s not easy to repair switching power supplys. I have reused some of them
where 5v dc are available frome other appliance.